bastibl/gr-ieee802-11 #6: wifi_phy_heir error: ieee802_11.wifi_signal_field() - Macports install

davidrpiper, 2015-01-08:
[OS X Yosemite 10.10.1, GNURadio v3.7.5.2]
From a fresh Macports (/opt/local/ path) installation I follow the instructions and try to open the wifi_phy_heir example.

It fails to run as the header_formatter variable value, ieee802_11.wifi_signal_field(), raises an error:

Param - Value(value):
	Value "ieee802_11.wifi_signal_field()" cannot be evaluated:
	name 'ieee802_11' is not defined

There is a mini-discussion of a workaround here:
https://www.ruby-forum.com/topic/4487026
However, it did not fix my problem. I implemented their fix as the following script (note it also fixes a bug for finding the log4cpp library):

# From a fresh Macports installation...
# Basics
set -x
sudo port install uhd 
sudo port install gnuradio +full

# For GNURadio
sudo port install log4cpp
sudo port install itpp

sudo port upgrade outdated

# GR-Foo (includes Wireshark module)
git clone https://github.com/bastibl/gr-foo.git
cd gr-foo
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/opt/local/

sed -i '' 's/\/usr\/bin\/python/\/opt\/local\/bin\/python2.7/g' CMakeCache.txt
sed -i '' 's/\/usr\/lib\/libpython2.7.dylib/\/opt\/local\/lib\/libpython2.7.dylib/g' CMakeCache.txt
sed -i '' 's/\/System\/Library\/Frameworks\/Python.framework\/Headers/\/opt\/local\/Library\/Frameworks\/Python.framework\/Versions\/2.7\/Headers/g' CMakeCache.txt

cmake .. -DCMAKE_INSTALL_PREFIX=/opt/local/

make
sudo make install

cd ../..

# GR-ieee802-11
git clone git://github.com/bastibl/gr-ieee802-11.git
cd gr-ieee802-11

sed -i '' 's/\${GNURADIO_RUNTIME_LIBRARY_DIRS}/\${GNURADIO_RUNTIME_LIBRARY_DIRS}\'$'\n    \/opt\/local\/lib/g' CMakeLists.txt
sed -i '' 's/\${ITPP_LIBRARIES}/\${ITPP_LIBRARIES}\'$'\n    log4cpp/g' lib/CMakeLists.txt

mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=/opt/local/

sed -i '' 's/\/usr\/bin\/python/\/opt\/local\/bin\/python2.7/g' CMakeCache.txt
sed -i '' 's/\/usr\/lib\/libpython2.7.dylib/\/opt\/local\/lib\/libpython2.7.dylib/g' CMakeCache.txt
sed -i '' 's/\/System\/Library\/Frameworks\/Python.framework\/Headers/\/opt\/local\/Library\/Frameworks\/Python.framework\/Versions\/2.7\/Headers/g' CMakeCache.txt

cmake .. -DCMAKE_INSTALL_PREFIX=/opt/local/

make
sudo make install

cd ../..

Anyone know/have a possible fix?

bastibl, 2015-01-08:
can you load the module in an interactive python session please.
python
>>> import ieee802_11
This should give you more debug output than running the flow graph in grc.

davidrpiper, 2015-01-08:
Output below - the second is using the Macports python.
```
$> python
Python 2.7.6 (default, Sep  9 2014, 15:04:36) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.39)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ieee802_11
>>> ^D

$> /opt/local/bin/python2.7
Python 2.7.9 (default, Dec 13 2014, 15:13:49) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.56)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ieee802_11
```

bastibl, 2015-01-08:
you have to adapt the python path to where the module is installed. something like
export PYTHONPATH=/opt/local/lib/python2.7/site-packages

davidrpiper, 2015-01-08:
Ah fantastic! Thank you so much! That solved it :)
```
$> export PYTHONPATH=/opt/local/lib/python2.7/site-packages
$> gnuradio-companion 
```

closed 2015-01-25 by bastibl

sharath35, 2015-06-30:
Executing: "/home/sharathbabu/gr-ieee802-11/examples/wifi_rx.py"

linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.008.000-6-gbde8e9a3

Traceback (most recent call last):
  File "/home/sharathbabu/gr-ieee802-11/examples/wifi_rx.py", line 32, in <module>
    import ieee802_11
ImportError: No module named ieee802_11

what is the solution for this?

bastibl, 2015-06-30:
adapt your PYTHONPATH. Python can't find the module.

sharath35, 2015-07-03:
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: "/home/sharathbabu/gr-ieee802-11/examples/wifi_loopback.py"

linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.008.000-6-gbde8e9a3

Using Volk machine: sse3_64
OFDM MAPPER: encoding: 0
set_min_output_buffer on block 31 to 96000
set_min_output_buffer on block 33 to 96000
set_min_output_buffer on block 35 to 96000
set_min_output_buffer on block 36 to 96000
set_min_output_buffer on block 39 to 96000
set_min_output_buffer on block 50 to 96000

how can i eliminate the warning?
i didn't see the 'Hello World' packets in console...

sharath35, 2015-07-03:
Executing: "/home/sharathbabu/gr-ieee802-11/examples/wifi_loopback.py"

linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.008.000-6-gbde8e9a3

Using Volk machine: sse3_64
OFDM MAPPER: encoding: 0
set_min_output_buffer on block 31 to 96000
set_min_output_buffer on block 33 to 96000
set_min_output_buffer on block 35 to 96000
set_min_output_buffer on block 36 to 96000
set_min_output_buffer on block 39 to 96000
set_min_output_buffer on block 50 to 96000
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 48000 KB
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::vmcircbuf_sysv_shm: shmget (1): Invalid argument
gr::buffer::allocate_buffer: failed to allocate buffer of size 48000 KB
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

what is bad_alloc ? please help me in rectifying this.

bastibl, 2015-07-03:
you can ignore the warning. Concerning the error, did you do this

https://github.com/bastibl/gr-ieee802-11#adjust-maximum-shared-memory

sharath35, 2015-07-07:
yeah i did... But what is the purpose of wifi loopback in this code? 
Please reply me with suitable answer or any related video.

sharath35, 2015-07-07:
i had used pc with ubuntu 14.04 and usrp n210 with an antenna and GB ethernet cable. That's all required or any required hardware is necessary?..

If so i am not getting the wireshark window ater executing wifi_transceiver.grc.

Kindly reply me with the reason.

bastibl, 2015-07-07:
The loopback flow graph is for loopback testing, without hardware. You will know that everything is installed correctly and that the basic things are working before sending anything over the air.
Over the air is generally much harder and there are several pitfalls. So it might make sense to test things locally as a start.
Apart from that can the flow graph easily be adapted for doing simulations and studying the performance of the system.

If you use the Wireshark Connector block you have to start Wireshark separately and read from the pipe you create in the flow graph (note file sink is in mode append). Look here to see how to start WIreshark

https://github.com/bastibl/gr-ieee802-11/blob/master/apps/rx_demo.sh#L43

Your hardware is fine, but first you will have to get the loopback flow graph working.

sharath35, 2015-07-23:
hello , 
when i execute the wifi_transceiver.grc file in gnuradio v3.7 . As i should get the output as the loopback, but i am getting the blank output in the graph. And continuosly getting DDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDDD......... infinite times. PFA
[image]

sharath35, 2015-07-23:
Can you suggest me how to get the output.... in the above case

bastibl, 2015-07-23:
...looks like your PC is too slow to process the incoming samples. 

http://files.ettus.com/manual/page_general.html#general_ounotes

sharath35, 2015-07-23:
ok thanks for your reply. Could you suggest me any modification to be done in the code to overcome these overflows....?

bastibl, 2015-07-23:
I guess the highest optimization potential has sync long and short

aler86, 2015-07-23:
Please, can you help me about wifi:loopback.grc?
when i run, the output is:
OFDM MAPPER: encoding: 0
set_min_output_buffer on block 31 to 96000
set_min_output_buffer on block 33 to 96000
set_min_output_buffer on block 35 to 96000
set_min_output_buffer on block 36 to 96000
set_min_output_buffer on block 39 to 96000
set_min_output_buffer on block 50 to 96000

I tried the instruction: 
sudo sysctl -w kernel.shmmax=2147483648

but nothing changes

aler86, 2015-07-23:
Now my output is

Showing: "/home/aler/gnuradio/gr-ieee802-11/examples/wifi_loopback.grc"

Generating: "/home/aler/gnuradio/gr-ieee802-11/examples/wifi_loopback.py"
>>> Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: "/home/aler/gnuradio/gr-ieee802-11/examples/wifi_loopback.py"

linux; GNU C++ version 4.6.3; Boost_104800; UHD_003.009.git-217-g012381d9

Using Volk machine: ssse3_32_orc
OFDM MAPPER: encoding: 0
set_min_output_buffer on block 31 to 96000
set_min_output_buffer on block 33 to 96000
set_min_output_buffer on block 35 to 96000
set_min_output_buffer on block 36 to 96000
set_min_output_buffer on block 39 to 96000
set_min_output_buffer on block 51 to 96000
thread[thread-per-block[13]: <block ofdm_mac (49)>]: pmt_cdr: wrong_type : xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx
 
but i don't see the scope diagram....can you help me please?

bastibl, 2015-07-23:
the first output you posted is not an error but information.

If you don't see the scope, it's most likely because you installed GNU Radio without WX GUI.

aler86, 2015-07-24:
I have gr-wxgui. How is it possible?

bastibl, 2015-07-24:
I don't know.... what exactly do you mean with "I don't see the scope diagram". The block is not there in GRC, the constellation does not appear, etc.

aler86, 2015-07-24:
i am getting the blank output in the graph....why?

bastibl, 2015-07-24:
....because no frame was detected and, thus, no constellations of carriers are shown

aler86, 2015-07-24:
can i get a frame?

aler86, 2015-07-24:
can i get a frame?

aler86, 2015-07-24:
can i get a frame?

aler86, 2015-07-24:
can i get a frame?