bastibl/gr-ieee802-11 #7: ImportError: dynamic module does not define init function (init_ieee802_11_swig)

rubund, 2015-01-25:
Hi,

When importing  ieee802_11 in python after a build and install, I get this error message:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 28, in <module>
    _ieee802_11_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_ieee802_11_swig', fp, pathname, description)
ImportError: dynamic module does not define init function (init_ieee802_11_swig)

I'm running Debian sid and using the latest version of gnuradio in the Debian archives (3.7.5-5).

Ruben

bastibl, 2015-01-25:
That might be due to an incompatible Swig version (I use 2.0.11). If that's not the problem can you please assert that python3 is _not_ used (actually cmake should take care of that).

rubund, 2015-01-25:
Thanks for the quick response. It's related to PythonLibs 3 vs. 2. I've created a pull request.

Thanks for the great effort you've put into gr-ieee802-11!

bastibl, 2015-01-25:
great, thank you for fixing this!

closed 2015-01-25 by bastibl