bastibl/gr-ieee802-11 #9: Undefined symbols for architecture x86_64

mycall, 2015-03-08:
How do you suggest fixing this?  I'm using latest OSX (with gnuradio working):

ndefined symbols for architecture x86_64:
  "log4cpp::CategoryStream::~CategoryStream()", referenced from:
      ofdm_decode_mac_impl::decode() in ofdm_decode_mac.cc.o
      ofdm_decode_signal_impl::print_signal() in ofdm_decode_signal.cc.o
      ofdm_parse_mac_impl::parse(boost::intrusive_ptr<pmt::pmt_base>) in ofdm_parse_mac.cc.o
      ofdm_sync_long_impl::general_work(int, std::__1::vector<int, std::__1::allocator<int> >&, std::__1::vector<void const*, std::__1::allocator<void const*> >&, std::__1::vector<void*, std::__1::allocator<void*> >&) in ofdm_sync_long.cc.o
      ofdm_sync_short_impl::insert_tag(unsigned long long) in ofdm_sync_short.cc.o
  "log4cpp::CategoryStream::operator<<(log4cpp::CategoryStream& (*)(log4cpp::CategoryStream&))", referenced from:
      ofdm_decode_mac_impl::decode() in ofdm_decode_mac.cc.o
      ofdm_decode_signal_impl::print_signal() in ofdm_decode_signal.cc.o
      ofdm_parse_mac_impl::parse(boost::intrusive_ptr<pmt::pmt_base>) in ofdm_parse_mac.cc.o
      ofdm_sync_long_impl::general_work(int, std::__1::vector<int, std::__1::allocator<int> >&, std::__1::vector<void const*, std::__1::allocator<void const*> >&, std::__1::vector<void*, std::__1::allocator<void*> >&) in ofdm_sync_long.cc.o
      ofdm_sync_short_impl::insert_tag(unsigned long long) in ofdm_sync_short.cc.o
  "log4cpp::eol(log4cpp::CategoryStream&)", referenced from:
      ofdm_decode_mac_impl::decode() in ofdm_decode_mac.cc.o
      ofdm_decode_signal_impl::print_signal() in ofdm_decode_signal.cc.o
      ofdm_parse_mac_impl::parse(boost::intrusive_ptr<pmt::pmt_base>) in ofdm_parse_mac.cc.o
      ofdm_sync_long_impl::general_work(int, std::__1::vector<int, std::__1::allocator<int> >&, std::__1::vector<void const*, std::__1::allocator<void const*> >&, std::__1::vector<void*, std::__1::allocator<void*> >&) in ofdm_sync_long.cc.o
      ofdm_sync_short_impl::insert_tag(unsigned long long) in ofdm_sync_short.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libgnuradio-ieee802_11.dylib] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [all] Error 2

bastibl, 2015-03-09:
Hmm, actually I thought log4cpp is optional now. Are you using the most recent version?
You could also fix this by installing liblog4cpp and recompile GNU Radio with logging.

bastibl, 2015-03-13:
Any news on that?

mycall, 2015-03-13:
osx doesn't have liblog4cpp in MacPorts.  I'll try updating and recompiling.

bastibl, 2015-03-15:
I guess it has

https://www.macports.org/ports.php?by=library&substr=log4cpp

and I already ran this code on OS X. Keep me updated.

bastibl, 2015-03-20:
Any news on that?

mycall, 2015-03-21:
Thanks for following up.  I haven't had time to recompile it.  I hope to review your ideas later.  Thanks!

closed 2015-03-21 by mycall

616e6f6e796d6f, 2015-08-31:
I've got the same error, recompiling with install of log4cpp doesn't help either...

[  9%] Linking CXX shared library libgnuradio-ieee802_11.dylib
Undefined symbols for architecture x86_64:
  "log4cpp::CategoryStream::~CategoryStream()", referenced from:
      ofdm_decode_mac_impl::decode() in ofdm_decode_mac.cc.o
      ofdm_decode_signal_impl::print_signal() in ofdm_decode_signal.cc.o
      ofdm_parse_mac_impl::parse(boost::intrusive_ptr<pmt::pmt_base>) in ofdm_parse_mac.cc.o
      ofdm_sync_short_impl::insert_tag(unsigned long long) in ofdm_sync_short.cc.o
  "log4cpp::CategoryStream::operator<<(log4cpp::CategoryStream& (*)(log4cpp::CategoryStream&))", referenced from:
      ofdm_decode_mac_impl::decode() in ofdm_decode_mac.cc.o
      ofdm_decode_signal_impl::print_signal() in ofdm_decode_signal.cc.o
      ofdm_parse_mac_impl::parse(boost::intrusive_ptr<pmt::pmt_base>) in ofdm_parse_mac.cc.o
      ofdm_sync_short_impl::insert_tag(unsigned long long) in ofdm_sync_short.cc.o
  "log4cpp::eol(log4cpp::CategoryStream&)", referenced from:
      ofdm_decode_mac_impl::decode() in ofdm_decode_mac.cc.o
      ofdm_decode_signal_impl::print_signal() in ofdm_decode_signal.cc.o
      ofdm_parse_mac_impl::parse(boost::intrusive_ptr<pmt::pmt_base>) in ofdm_parse_mac.cc.o
      ofdm_sync_short_impl::insert_tag(unsigned long long) in ofdm_sync_short.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libgnuradio-ieee802_11.dylib] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [all] Error 2

bastibl, 2015-09-01:
grr, this logging stuff causes lots of gray hairs... I just pushed a commit that hopefully fixes your issue.

616e6f6e796d6f, 2015-09-01:
Dude within a day?!?! You rock... My ISP should learn from guys like you!
Do you think the wifi_rx.grc should work with osmocom source?

bastibl, 2015-09-01:
Great that it works for you.
I don't have any hardware that I could try with the osmocom source, but I see no reason why it should not work.

616e6f6e796d6f, 2015-09-01:
I've just tried it out on a hackrf and it runs.. But I'm getting overruns. The only thing I've changed is the freq which I've set tot freq - lo_offset. 
Using Volk machine: avx_64_mmx_orc
gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.8
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy 
Using HackRF One with firmware 2014.08.1 
OOOOOOOgr::log :INFO: ofdm_decode_signal0 - encoding: 0 - length: 478 - symbols: 161
OOOOOOOOOOOOOOOOOOOOOOOOOOOOOOgr::log :INFO: ofdm_decode_signal0 - encoding: 4 - length: 779 - symbols: 66
OOOOOOOOOOOOOOOOOOgr::log :INFO: ofdm_decode_signal0 - encoding: 2 - length: 1334 - symbols: 223
OOgr::log :INFO: ofdm_decode_signal0 - encoding: 7 - length: 2611 - symbols: 97
Ogr::log :INFO: ofdm_decode_signal0 - encoding: 4 - length: 779 - symbols: 66
OOOOOOOgr::log :INFO: ofdm_decode_signal0 - encoding: 1 - length: 1454 - symbols: 324
OOOOOOOOOOOOOOOOgr::log :INFO: ofdm_decode_signal0 - encoding: 0 - length: 582 - symbols: 195
OOOOOOOOOgr::log :INFO: ofdm_decode_signal0 - encoding: 0 - length: 582 - symbols: 195

bastibl, 2015-09-01:
You could try running ```volk-profile```. If you still experience that much overruns you should get a faster PC :-/

616e6f6e796d6f, 2015-09-01:
Hmm run volk last night. And I'm running on a brand new macpro... So that couldn't be the problem (I hope)

bastibl, 2015-09-01:
Maybe you should assure that Sync Short does not detect frames on silent channels.
Maybe you see a DC offset that triggers frame detection and floods the rest of the flow graph with noise samples.

616e6f6e796d6f, 2015-09-02:
I have a large DC spike in my fft... Will try a dc blocker, I'll let you know.

616e6f6e796d6f, 2015-09-02:
Crap... in all try and error I only once got 1 partial packet in the pcap.

giladbeeri, 2016-11-28:
I had a problem with linking to log4cpp (which was installed) when building the GNU Radio tutorial (not gr-ieee802-11) and I solved it by running "CC=clang CXX=clang cmake .." instead of "cmake ..".