bastibl/gr-ieee802-11 #16: Unable to import gr-foo (or gr-ieee802-11) on my Mac

xuancaishaun, 2015-06-23:
The UHD and GNURadio versions on my Mac are as follows:
```
$ port installed gnuradio
The following ports are currently installed:
  gnuradio @3.7.6.1_2+docs+grc+jack+orc+portaudio+qtgui+sdl+swig+uhd+wavelet+wxgui
```
```
$ port installed uhd
The following ports are currently installed:
  uhd @3.8.2_1+docs+examples+libusb+manpages+manual+orc+test
  uhd @3.8.4_0+docs+examples+libusb+manpages+manual+orc+test (active)
```

The $PATHONPATH is as follows:
```
$ echo $PATHONPATH
/usr/local/lib/python2.7/site-packages/
```
Error occurred when I tried to import foo in Python:
```
Python 2.7.10 (default, May 25 2015, 13:06:17) 
[GCC 4.2.1 Compatible Apple LLVM 6.0 (clang-600.0.56)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import foo
Mac OS; Clang version 6.1.0 (clang-602.0.53); Boost_105800; UHD_003.008.004-MacPorts-Release

Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6
```

Similar error occurred when importing ieee802_11 in Python.

Error details are as follows:
```
Process:               Python [8159]
Path:                  /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python
Identifier:            Python
Version:               2.7.10 (2.7.10)
Code Type:             X86-64 (Native)
Parent Process:        bash [7337]
Responsible:           Terminal [3939]
User ID:               501

Date/Time:             2015-06-22 20:16:56.546 -0500
OS Version:            Mac OS X 10.10.3 (14D136)
Report Version:        11
Anonymous UUID:        AE8F3DCD-972A-7BF2-BD0E-47629E93CB83

Sleep/Wake UUID:       D061BE21-53B5-49B4-9E08-490F812363CE

Time Awake Since Boot: 220000 seconds
Time Since Wake:       2200 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff91caf286 __pthread_kill + 10
1   libsystem_c.dylib             	0x00007fff93bf0b53 abort + 129
2   org.python.python             	0x00000001022dca57 Py_FatalError + 49
3   org.python.python             	0x00000001022dbf39 PyThreadState_Get + 28
4   org.python.python             	0x00000001022d8d96 Py_InitModule4_64 + 62
5   _foo_swig.so                  	0x0000000102188c50 init_foo_swig + 816
6   org.python.python             	0x0000000101c8647c _PyImport_LoadDynamicModule + 140
7   org.python.python             	0x0000000101c84a29 imp_load_module + 297
[... 147 lines cut ...]
VM_ALLOCATE                           8K
__DATA                             2836K
__LINKEDIT                         78.8M
__TEXT                             51.1M
__UNICODE                           552K
shared memory                         4K
===========                      =======
TOTAL                             227.5M

Model: MacBookPro11,1, BootROM MBP111.0138.B14, 2 processors, Intel Core i5, 2.4 GHz, 8 GB, SMC 2.16f68
Graphics: Intel Iris, Intel Iris, Built-In
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, -
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x02FE, -
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x112), Broadcom BCM43xx 1.0 (7.15.166.24.3)
Bluetooth: Version 4.3.4f4 15601, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: Internal Memory Card Reader
USB Device: USB Gaming Mouse
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 17.2

```

bastibl, 2015-06-23:
It's $PYTHONPATH, not $PATHONPATH.
Apart from that you have to link everything against the port python libraries. Currently, it looks like you compiled some stuff against MAC python and some against ports python.
Use cmake to configure the ptyhon libraries. (I don't have ports installed so I cant tell you the exact path, but I guess its somewhere under /oprt/Libraries

xuancaishaun, 2015-06-24:
Thanks for your response. But I found it very troublesome to fix the issue on Mac at this moment. I got it running on my Fedora laptops anyway. I will come back and review your suggestion.

closed 2015-06-24 by xuancaishaun

bastibl, 2015-06-25:
OK, I'm happy it works on Linux. Btw, I think gr-ieee802-11 is also in ports. Maybe you want to give this version a try.

xuancaishaun, 2015-06-26:
So I managed to resolve the issue by switching to the Python version on Mac instead of the one installed by MacPort. 

```
$ port select --list python
Available versions for python:
	none (active)
	python26-apple
	python27
	python27-apple 
```

Then select "python27-apple" as the default Python:
```
$ sudo port select python python27-apple
Selecting 'python27-apple' for 'python' succeeded. 'python27-apple' is now active.
```

Exclude "PYTHONPATH" in ~/.bash_profile and recompile the libraries.

bastibl, 2015-06-26:
great, thanks for reporting back on this.

xuancaishaun, 2015-06-28:
Here are some more updates, since I realized I didn't solve the problem last time. 

The output of cmake is the following:
```
$ sudo cmake ..
-- The CXX compiler identification is AppleClang 6.1.0.6020053
-- The C compiler identification is AppleClang 6.1.0.6020053
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found PkgConfig: /opt/local/bin/pkg-config (found version "0.28")
-- checking for module 'uhd'
--   found uhd, version 003.009.git-MacPorts-git-51bc00ee(20150511)
-- Found UHD: /opt/local/lib/libuhd.dylib 
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
Checking for GNU Radio Module: RUNTIME
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.8git
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_RUNTIME: /opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib 
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.8git
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
-- Found GNURADIO_PMT: /opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib 
GNURADIO_PMT_FOUND = TRUE
-- checking for module 'cppunit'
--   found cppunit, version 1.13.2
-- Found CPPUNIT: /opt/local/lib/libcppunit.dylib 
CMake Warning (dev) at /opt/local/share/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "test-foo".  Use the
  target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
  lib/CMakeLists.txt:80 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

--
-- Checking for module SWIG
-- Found SWIG version 3.0.5.
-- Found SWIG: /opt/local/bin/swig 
-- Found PythonLibs: /usr/lib/libpython2.7.dylib (found suitable version "2.7.6", minimum required is "2")
-- Found PythonInterp: /opt/local/bin/python2.7 (found suitable version "2.7.10", minimum required is "2")
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE
-- Performing Test HAVE_WNO_UNUSED_BUT_SET_VARIABLE - Failed
-- Found Doxygen: /opt/local/bin/doxygen (found version "1.8.9.1")
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/shaun/bitbucket/gr-foo/build
```

Notice that **PythonLibs** found by cmake is the one at `/usr/lib/libpython2.7.dylib`, which, however, is NOT the one used by MacPort and GNURadio. It is the root cause for the above issue.

In Mac, I think there are (at least) three types of Pythons (for the same version, say, 2.7). 
* The one installed by Mac itself: `/Library/Frameworks/Python.framework/Versions/2.7`
* The one installed by users (possibly through Python installers):`/usr/lib/python2.7`
* The one installed by MacPort: `/opt/local/Library/Frameworks/Python.framework/Versions/2.7`

To compile and import gr-foo in GNURadio, we need to find the correct **PythonLibs**, which should be `/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib`.

So the next step is to change the corresponding paths in cmake. The approach (ref: http://stackoverflow.com/questions/7660001/cmake-finds-wrong-python-libs) I was using is the following:
```
$cd gr-foo/build
$ccmake ..
```

Then press `c` to configure, and `t` for advanced options. Change PythonLibs to the correct one. Actually, we may change all `/usr/local/...` to `/opt/local/...` to be consistent, including cmake installing paths. 

Finally, make sure `PYTHONPATH` in `~/.bash_profile` is set:
```
export PYTHONPATH="/opt/local/lib/python2.7/site-packages/:$PYTHONPATH"
```
Type `source ~/.bash_profile` to make it effective.

bastibl, 2015-06-29:
Yes, that's exactly what I meant with: you have to link everything against the port python libraries