bastibl/gr-ieee802-11 #19: swig problem: 'Import "import ieee802_11" failed.'

jiska2342, 2015-09-01:
Hi,

i'm running gnuradio version 3.7.5-5 installed from Debian testing. Building gr-foo and gr-ieee802-11 works without errors. When importing the ieee802_11 module from python, it is working:

```
$ python -v
...
>>> import ieee802_11
import ieee802_11 # directory /usr/lib/python2.7/dist-packages/ieee802_11
# /usr/lib/python2.7/dist-packages/ieee802_11/__init__.pyc matches /usr/lib/python2.7/dist-packages/ieee802_11/__init__.py
import ieee802_11 # precompiled from /usr/lib/python2.7/dist-packages/ieee802_11/__init__.pyc
# /usr/lib/python2.7/plat-x86_64-linux-gnu/DLFCN.pyc matches /usr/lib/python2.7/plat-x86_64-linux-gnu/DLFCN.py
import DLFCN # precompiled from /usr/lib/python2.7/plat-x86_64-linux-gnu/DLFCN.pyc
# /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.pyc matches /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py
import ieee802_11.ieee802_11_swig # precompiled from /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.pyc
import imp # builtin
dlopen("/usr/lib/python2.7/dist-packages/ieee802_11/_ieee802_11_swig.so", 102);
import _ieee802_11_swig # dynamically loaded from /usr/lib/python2.7/dist-packages/ieee802_11/_ieee802_11_swig.so
>>> ieee802_11.BPSK_1_2
0
>>> 
```

Yet, when using it in gnuradio, it keeps telling me that the import failed. When opening the wifi_phy_hier.grc I get a couple of errors like this:

```
Error 3:
Block - encoding - Parameter(parameter):
  Param - Value(value):
    Value "ieee802_11.BPSK_1_2" cannot be evaluated:
    name 'ieee802_11' is not defined
```

For further debugging, I built a minimal gnuradio example consisting of a simple import block for the module. To verify the module is actually loading etc. I started gnuradio like this:

```
$ /usr/bin/python -v //usr/bin/gnuradio-companion
...
import ieee802_11 # directory /usr/lib/python2.7/dist-packages/ieee802_11
# /usr/lib/python2.7/dist-packages/ieee802_11/__init__.pyc matches /usr/lib/python2.7/dist-packages/ieee802_11/__init__.py
import ieee802_11 # precompiled from /usr/lib/python2.7/dist-packages/ieee802_11/__init__.pyc
# /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.pyc matches /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py
import ieee802_11.ieee802_11_swig # precompiled from /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.pyc
dlopen("/usr/lib/python2.7/dist-packages/ieee802_11/_ieee802_11_swig.so", 102);
import ieee802_11 # directory /usr/lib/python2.7/dist-packages/ieee802_11
# /usr/lib/python2.7/dist-packages/ieee802_11/__init__.pyc matches /usr/lib/python2.7/dist-packages/ieee802_11/__init__.py
import ieee802_11 # precompiled from /usr/lib/python2.7/dist-packages/ieee802_11/__init__.pyc
# /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.pyc matches /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py
import ieee802_11.ieee802_11_swig # precompiled from /usr/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.pyc
dlopen("/usr/lib/python2.7/dist-packages/ieee802_11/_ieee802_11_swig.so", 102);
Showing: "/tmp/untitled.grc"
```

However, the import block keeps telling me:

```
	Param - Import(import):
		Import "import ieee802_11" failed.
```

For further debugging, I was hand-editing the module files. It turned out that the import error in gnuradio only occurs if _ieee802_11_swig.so is imported - which doesn't raise any error on the python command line. However, without _ieee802_11_swig.so, all the contents of the module are missing ;)

A screenshot of my minimal error example can be found here:
[image]

Do you have any idea how I can get rid of this import error? The python path is correct and all modules are actually found...

Best,
Jiska

bastibl, 2015-09-01:
I've never seen this error and, unfortunately, I don't have an explanation for it.

Sometime GRC hides the actual error. Any chances it's the LD_LIBRARY_PATH (instead of the PYTHONPATH) which is not set correctly in the GRC environment?

Maybe you could also try ```sudo ldconfig```...

jiska2342, 2015-09-01:
I did ldconfig and also set the LD_LIBRARY_PATH... I'll try to reproduce this error in a virtual machine and/or with another gnuradio version later.

Do you have any idea how i could get more output from GRC?

bastibl, 2015-09-01:
I only have wild guesses :-/

Maybe you use different python versions? Is the version GRC starts different to what is sym-linked by /usr/bin/python?

I moved libgnuradio-ieee802-11.so out of my LD_LIBRARY_PATH and it triggered your error.

If you put something like

```
import sys; print sys.path
```

or 
```
import os; os.environ
```
in the import statement you can check GRC variables.

jiska2342, 2015-09-01:
I also have gr-foo installed in the same path. For this I don't get any errors, so I really think it is no path problem. Python versions are the same and the output from your commands looks good.

bastibl, 2015-09-01:
how does ```/usr/include/gnuradio/config.h``` look like?

Also could you please post the output  from
```
ldd /usr/lib/libgnuradio-runtime.so
ldd /usr/lib/libgnuradio-ieee802-11.so
```

jiska2342, 2015-09-01:
I could reproduce the error as follows:

I started Kali Linux in vitualbox (the most current version from http://cdimage.kali.org/kali-2.0/kali-linux-2.0-amd64.iso).

I installed the following packets: ```cmake libuhd-dev libboost-thread-dev libboost-filesystem-dev libboost-system-dev libcppunit-dev gnuradio-dev swig doxygen liblog4cpp5-dev libitpp-dev```

I used the same commands as shown on the website except from the following ```cmake -DCMAKE_INSTALL_PREFIX=/usr ..```

...and now get the same error as described. I guess I should not change the installation prefix but change all paths.

jiska2342, 2015-09-01:
```# cat /usr/include/gnuradio/config.h
/*
 * Copyright 2013 Free Software Foundation, Inc.
 *
 * This file is part of GNU Radio
 *
 * GNU Radio is free software; you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation; either version 3, or (at your option)
 * any later version.
 *
 * GNU Radio is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with GNU Radio; see the file COPYING.  If not, write to
 * the Free Software Foundation, Inc., 51 Franklin Street,
 * Boston, MA 02110-1301, USA.
 */

#ifndef GNURADIO_CONFIG_H
#define GNURADIO_CONFIG_H
#ifndef TRY_SHM_VMCIRCBUF
#define TRY_SHM_VMCIRCBUF
#endif
#ifndef GR_PERFORMANCE_COUNTERS
#define GR_PERFORMANCE_COUNTERS
#endif
#ifndef GR_CTRLPORT
#define GR_CTRLPORT
#endif
#ifndef ENABLE_GR_LOG
#define ENABLE_GR_LOG
#endif
#ifndef HAVE_LOG4CPP
#define HAVE_LOG4CPP
#endif

#endif /* GNURADIO_CONFIG_H */```

```
# ldd /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so
        linux-vdso.so.1 (0x00007ffeeefad000)
        libvolk.so.0.0.0 => /usr/lib/x86_64-linux-gnu/libvolk.so.0.0.0 (0x00007f6c3dce3000)
        libgnuradio-pmt.so.3.7.5 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.5 (0x00007f6c3da94000)
        libboost_date_time.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.55.0 (0x00007f6c3d881000)
        libboost_program_options.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.55.0 (0x00007f6c3d609000)
        libboost_filesystem.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.55.0 (0x00007f6c3d3f1000)
        libboost_system.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.55.0 (0x00007f6c3d1ec000)
        libboost_thread.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.55.0 (0x00007f6c3cfd1000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f6c3cdb4000)
        liblog4cpp.so.5 => /usr/lib/liblog4cpp.so.5 (0x00007f6c3cb70000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f6c3c968000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f6c3c764000)
        liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f6c3c4dd000)
        libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f6c3c1d1000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f6c3bed0000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f6c3bcb9000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f6c3b910000)
        /lib64/ld-linux-x86-64.so.2 (0x000056240981a000)
        libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007f6c3b6f7000)
```

```
# ldd /usr/lib/libgnuradio-ieee802_11.so 
        linux-vdso.so.1 (0x00007fff76b3b000)
        libboost_filesystem.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.55.0 (0x00007f1bf2490000)
        libboost_system.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.55.0 (0x00007f1bf228c000)
[... 10 lines cut ...]
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1bf019b000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1beff7e000)
        libvolk.so.0.0.0 => /usr/lib/x86_64-linux-gnu/libvolk.so.0.0.0 (0x00007f1befc74000)
        libboost_date_time.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_date_time.so.1.55.0 (0x00007f1befa62000)
        libboost_program_options.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.55.0 (0x00007f1bef7ea000)
        libboost_thread.so.1.55.0 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.55.0 (0x00007f1bef5ce000)
        liblog4cpp.so.5 => /usr/lib/liblog4cpp.so.5 (0x00007f1bef38b000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f1bef187000)
        liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f1beef00000)
        libgnuradio-blocks.so.3.7.5 => /usr/lib/x86_64-linux-gnu/libgnuradio-blocks.so.3.7.5 (0x00007f1beeab1000)
        libgnuradio-analog.so.3.7.5 => /usr/lib/x86_64-linux-gnu/libgnuradio-analog.so.3.7.5 (0x00007f1bee83d000)
        libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007f1bee430000)
        libfftw3f_threads.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3 (0x00007f1bee229000)
        libfftw3.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3.so.3 (0x00007f1bede27000)
        liblapack.so.3 => /usr/lib/liblapack.so.3 (0x00007f1bed62b000)
        libblas.so.3 => /usr/lib/libblas.so.3 (0x00007f1bed3ec000)
        libgomp.so.1 => /usr/lib/x86_64-linux-gnu/libgomp.so.1 (0x00007f1bed1ca000)
        /lib64/ld-linux-x86-64.so.2 (0x0000555e377c8000)
        libnsl.so.1 => /lib/x86_64-linux-gnu/libnsl.so.1 (0x00007f1becfb1000)
        libatlas.so.3 => /usr/lib/libatlas.so.3 (0x00007f1beca13000)
        libgfortran.so.3 => /usr/lib/x86_64-linux-gnu/libgfortran.so.3 (0x00007f1bec6f2000)
        libquadmath.so.0 => /usr/lib/x86_64-linux-gnu/libquadmath.so.0 (0x00007f1bec4b3000)
```

Linking /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so to /usr/lib/libgnuradio-runtime.so did not help.

bastibl, 2015-09-01:
Maybe it's related to log4cpp. It's the only library which is linked to /usr/lib. Maybe it's the 32-bit version.
But in that case also GNU Radio would be linked incorrectly. Are any flow graphs working for you (like from the example directories of GNU Radio).

jiska2342, 2015-09-01:
Bad news, it must be something different. I just did ```cmake ..``` on my current system which installed it to /usr/local, and set all paths. It's working in python but not in GRC.

Something that is working for me is your ```wifi_rx.grc``` :) It has no ```ieee802_11``` python reference. I changed the source from that yesterday and got it running with my rad1o badge (almost similar to HackRF).

bastibl, 2015-09-01:
...and you are exporting your variables with
```
export LD_...=
```
and not
```
LD_...=
```
and you are starting gnuradio-companion from that very same terminal?

jiska2342, 2015-09-01:
```
export LD_LIBRARY_PATH=/usr/local/lib
export PYTHONPATH=/usr/local/lib/python2.7/dist-packages
 /usr/bin/python -v /usr/bin/gnuradio-companion
```
In the config, I set ```local_blocks_path = /usr/local/share/gnuradio/grc/blocks```.

The path is now pointing to /usr/local since I just used ``cmake ..`` - and I am getting the same error.

I'll try another gnuradio version, since Debian testing and Kali Linux have the same.

bastibl, 2015-09-01:
I downloaded the image and added debug output to the import. The problem is

```
dlopen: cannot load any more object with static TLS
```

which seems to be caused by a rather small compile time setting of Kalis glibc. You can work around by preloading the library.

```
export LD_PRELOAD=/usr/local/lib/libgnuradio-ieee802-11.so
```

jiska2342, 2015-09-01:
Awesome :) That solved this problem for both, Debian testing and Kali Linux.

bastibl, 2015-09-01:
Great :-)

closed 2015-09-01 by bastibl