bastibl/gr-ieee802-11 #20: libgnuradio-ieee802_11.so not found?

egaebel, 2015-09-24:
Hi, I've been setting up gr-ieee802-11 on a Linux Mint 17 installation with gnuradio 3.7.3.

Right now I'm just trying to build "/examples/wifi_phy_hier.grc" as detailed on the main Github page. However I get the following errors in Gnuradio-companion:

Error 0:
Block - header_formatter - Variable(variable):
  Param - Value(value):
    Value "ieee802_11.wifi_signal_field()" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 1:
Block - ieee802_11_ofdm_mapper_0 - OFDM Mapper(ieee802_11_ofdm_mapper):
  Check "$encoding in range(8)" did not evaluate.

Error 2:
Block - ieee802_11_ofdm_mapper_0 - OFDM Mapper(ieee802_11_ofdm_mapper):
  Param - Encoding(encoding):
    Value "encoding" cannot be evaluated:
    name 'encoding' is not defined

Error 3:
Block - encoding - Parameter(parameter):
  Param - Value(value):
    Value "ieee802_11.BPSK_1_2" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 4:
Block - digital_packet_headergenerator_bb_0 - Packet Header Generator(digital_packet_headergenerator_bb):
  Param - Formatter Object(header_formatter):
    Value "header_formatter.formatter()" cannot be evaluated:
    name 'header_formatter' is not defined

So after some googling I saw people suggesting to see the result of importing ieee802_11 in an interactive python session. This fails with the following import error:

>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/opt/local/lib/python2.7/dist-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
  File "/opt/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 28, in <module>
    _ieee802_11_swig = swig_import_helper()
  File "/opt/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_ieee802_11_swig', fp, pathname, description)
ImportError: libgnuradio-ieee802_11.so: cannot open shared object file: No such file or directory

So I went searching for libgnuradio-ieee802_11.so and found it in /opt/local/lib.

What really gets me is I don't know why the line:

    _mod = imp.load_module('_ieee802_11_swig', fp, pathname, description)

is even associated with libgnuradio-ieee802_11.so, since that isn't the file it's trying to load. I suppose the _ieee802_11_swig.so file must call it?

I also tried putting a copy of libgnuradio-ieee802_11.so in the /opt/local/lib/python2.7/dist-packages/ieee802_11/ folder to see if that would work with no luck.

I'm guessing there is something I did wrong in setting everything up, but I can't figure out what. Any ideas?

bastibl, 2015-09-24:
Hi,
please put ```/opt/local/lib``` in your LD_LIBRARY_PATH environment variable and do ```sudo ldconfig```.

egaebel, 2015-09-24:
That did it. I feel silly. Thank you so much!

bastibl, 2015-09-24:
Good to hear that it's fixed

closed 2015-09-24 by bastibl

aler86, 2015-10-16:
I found your work very interesting and i have tried of modify your code for implementig 256-QAM, but i have trouble.
Can you help me about 256 QAM, it's possibile to update this modulation?

tagsys, 2017-05-15:
I encountered the same issue, but on MAC. There is no LD_LIBRARY_PATH environment variable on MAC. How could I resolve this issue? Many thanks in advance.

bastibl, 2017-05-15:
On macOS you should not need that. The shared library should know the full install path of the linked libraries. If you have problems, the library that cannot be found probably hasn't set its install_name.

CamiloCaicedo, 2017-10-12:
Hello Friend.
I am also installing gr-ieee802-11 following the steps of the Github page. I'm working with Linux Mint 18.1 Serena x64 bit and gnuradio 3.7.11.1. When I open the file wifi_phy_hier.grc to build the block, I get the following errors:

Block - header_formatter - Variable(variable):
  Param - Value(value):
    Value "ieee802_11.signal_field()" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 1:
Block - chan_est - Parameter(parameter):
  Param - Value(value):
    Value "ieee802_11.LS" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 2:
Block - digital_packet_headergenerator_bb_0 - Packet Header Generator(digital_packet_headergenerator_bb):
  Param - Formatter Object(header_formatter):
    Value "header_formatter.formatter()" cannot be evaluated:
    name 'header_formatter' is not defined

Error 3:
Block - encoding - Parameter(parameter):
  Param - Value(value):
    Value "ieee802_11.BPSK_1_2" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 4:
Block - ieee802_11_frame_equalizer_0 - WiFi Frame Equalizer(ieee802_11_frame_equalizer):
  Param - Algorithm(algo):
    Value "chan_est" cannot be evaluated:
    name 'chan_est' is not defined

Error 5:
Block - ieee802_11_mapper_0 - WiFi Mapper(ieee802_11_mapper):
  Check "$encoding in range(8)" did not evaluate.

Error 6:
Block - ieee802_11_mapper_0 - WiFi Mapper(ieee802_11_mapper):
  Param - Encoding(encoding):
    Value "encoding" cannot be evaluated:
    name 'encoding' is not defined

I have tried to do several things I have read in some forums but the errors keep popping up. Could you help me solve them?

CamiloCaicedo, 2017-10-12:
That's how it looks GNURADIO:

[image]

bastibl, 2017-10-13:
As you might have seen, the problem was that the `LD_LIBRARY_PATH` was not set up. Did you check that yours is correct?

If that's not the problem, you might try opening a Python session and `import ieee802_11`.

CamiloCaicedo, 2017-10-18:
with respect to LD_LIBRARY_CONFIG, I do the following:

export LD_LIBRARY_PATH=/usr/local/lib

I also opened the session of python and import ieee802_11. the result was:

Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ieee802_11

I read that I should add the module installation directory to the Python path
 
exportar PYTHONPATH = / opt / local / lib / python2.7 / site-packages

but, when I start a python session again and import ieee802_11 I get the same result.

>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named ieee802_1

bastibl, 2017-10-18:
It seems you are copy-pasting random commands you find online. First, you have to find out where you actually installed the module. See if the library is actually under `/usr/local/lib`.

If that's the case, you could setup the environment with:
```
export LD_LIBRARY_PATH=/usr/local/lib:${LD_LIBRARY_PATH}
export PYTHONPATH=/usr/local/lib/python2.7/site-packages:${PYTHONPATH}
```

Also run `sudo ldconfig`.

CamiloCaicedo, 2017-10-18:
I am new to the subject. Do you mean where to install gr-ieee802-11? or which module do you mean?

bastibl, 2017-10-19:
OK, so maybe we should go 10 steps back.
- Did you download, build, and install gr-ieee802-11?
- If so, where did you install it to?

I understand that my previous post is hard to understand, but did you actually try the commands?

CamiloCaicedo, 2017-10-19:
[image]

as you can see in the image, I download gr-ieee802_11 in / home / camilo. I had to install it on another route? 

Yes, probe with:
export LD_LIBRARY_PATH=/usr/local/lib:${LD_LIBRARY_PATH}
export PYTHONPATH=/usr/local/lib/python2.7/site-packages:${PYTHONPATH}

but but the problem continues.

bastibl, 2017-10-20:
OK, so you downloaded the source code. That's already pretty good. The next important steps are to actually compile and install the software. Since there is already a build directory, you seem to (at least) have tried something. 

Please see here for installation instructions: https://www.wime-project.net/installation/
Don't just blindly copy-paste commands to the terminal and hope for the best, but make sure that it actually works. During installation, you will also see where the module ends up. I don't have Mint, so I don't know the default install paths on Mint. Then adapt your environment variables accordingly.

I'm not really sure how I can help you further with this. What you post, suggests that you are missing the basics of using a computer. Maybe you want to read a bit more about Linux and compiling software. The fact that you posted a screenshot of the downloaded source code, suggests that you don't know the difference between source code, build directory, and install path.