bastibl/gr-ieee802-11 #21: OFDM 256-QAM

aler86, 2015-10-16:
Good Morning.
I found your work very interesting and i have tried of modify your code for implementig 256-QAM, but i have trouble.
Can you help me about 256 QAM, it's possibile to update this modulation?

aler86, 2015-10-16:
actually, my problem it is in file utils.cc in rate_field. how can I define two other fields modulation?

bastibl, 2015-10-17:
Since 256QAM is not part of the standard there is no /right/ way to add this modulation. Just use some bits in the signal field.

aler86, 2015-10-19:
I would  add this modulation, because I'm trying  to adapt your code for standard IEE 802.11 ac.

Do you think it is possible?

Thank you.

bastibl, 2015-10-19:
Depends on what you are trying to achieve. It's certainly possible to implement 11ac, but I doubt you will manage 160MHz over the air.

aler86, 2015-10-19:
Ok thank you,

Do you think it is possible manage 80MHz over the air?

As soon as I add the 256 QAM. Can I do a git commit in order to contribute at  your code?

aler86, 2015-10-19:
Please can you help me,
In file Utils.cc, and in particular in function "void puncturing", How  can I add a 256QAM with code rate 5/6?

bastibl, 2015-10-20:
I'm afraid you have to be way more specific in what's your problem. Actually you should checkout the code rates defined in the standard, lookup their puncturing patterns, and add them.

But apart from that, I think the main problem is that 11ac is way, way different from 11a/g/p and a straight forward adaption is not easily possible. There are different FEC algorithms, preamble formats, MIMO modes, etc...

aler86, 2015-10-21:
Oh, I see.I think it should first adapt in 802.11n and then in 802.11 ac.
Would you like to help me in this challenge?

Have you C\C++ code for phy and mac 802.11 n or 802.11 ac?
Thank you

bastibl, 2015-10-21:
Unfortunately, I don't have the time to work on 11n/11ac support at the moment.
It would, of course, be great if you work on adding support for these versions of the standard.
Once you have a specific technical question feel free to ask.

closed 2015-10-21 by bastibl