bastibl/gr-ieee802-11 #22: Submissions Upstream

meeas, 2015-10-21:
Great project!  I have yet to try compiling it yet, but have you tried submitting any of your blocks to the GNURadio project?  Once like your 802.11 and Wireshark blocks?

bastibl, 2015-10-22:
Hi, thanks for your interest. No, I didn't submit anything upstream. GNU Radio wants to be an SDR framework that provides an environment to implement stuff, but that's free from applications and implementations of specific technologies; and I think that's a good idea.
For applications there is the pybombs app-store and the cgran website.
Tom described that pretty good in this panel discussion https://www.youtube.com/watch?v=m1t_66gafOQ

closed 2015-10-22 by bastibl

fujiwaraemi, 2015-11-11:
Hello, I'm very interested in your research and l have downloaded the codes of gr-802-11 for researching WiFi over USRP! But when I want to compile the first step on "cmake .." of gr-foo followed the README, the terminal pointed out "Could not find UHD (missing: UHD_LIBRARIES)" "CMake Error at CMakeLists.txt:48(message): UHD library not find" . Now I have installed the UHD of 3.7.1, is there a problem with the version of UHD? Which version of UHD is required? Looking forward to your reply. Thank you!

bastibl, 2015-11-11:
Hi, please install the Ettus UHD driver. It's the driver for your USRP and you will need it either way. See https://github.com/EttusResearch/uhd

aler86, 2015-11-11:
Hi i have a question about gr-802.11,
is it possible implement this tool through Hackrf one instead UHD?

Thanks in advance

fujiwaraemi, 2015-11-12:
Thank you for your useful advise, I've solved the problem with UHD! But when I want to check my installation executing  "wifi_loopback.grc" as README, I could not  see some decoded 'Hello World' packets in the console but a series of letter "x". Also when I changed the modulate types as the followed pictures, the plots maybe the subcarrier constellations, however they are different from the theoretical constellation diagram. Is there something wrong with my installation?
[image]
[image]
[image]

bastibl, 2015-11-12:
Hi,

I would recommend to install python-opengl, then your graphs will look much nicer. What you currently  see is the non-gl version.

The 'X's are the payload of the frame. It's not hello world anymore. Some time ago we made the PDU length changeable through the slider and just fed in a given number of 'X's (I guess that should be updated in the readme). So that's perfectly fine. Use the 'PDU length' slider to create shorter packets and you will see also other output in your console (you could also scroll up)