bastibl/gr-ieee802-11 #23: Implementation gr-802.11 to Hackrf one Hardware

aler86, 2015-11-11:
Hi i have a question about gr-802.11,
is it possible implement this tool through Hackrf one instead UHD?

Thanks in advance

bastibl, 2015-11-11:
Hi, yes that's possible. Just replace the UHD Sink / Source with the Osmocom blocks.

closed 2015-11-11 by bastibl

aler86, 2015-11-11:
Yes, i know it.
 But, according to you, Hackrff one It is able to manage  standard 802.11?
Do I need other hardware?

bastibl, 2015-11-11:
...depends on what you mean with "manage standard 802.11".

This module implements the physical layer including all framing to generate and receive standard compliant frames. It does //not// include timing related MAC layer functionality like backoffs. Furthermore, this architecture will not meet all timing requirements of the standard (like sending back ACKs in time)

aler86, 2015-11-11:
Ok thanks, i update you as soon as possible...