bastibl/gr-ieee802-11 #26: Correlation Output Initialization?

bkihei, 2016-01-06:
I noticed that the **autocorrelation output** in the receiver chain appears to have an initialization phase (oscillator stabilization?) that expands above .56, and doesn't settle until about 1e7 samples. The other high spikes are packets that were being detected from a COTS 802.11p radio.

[image]

I added some some code ofdm_sync_short.cc to tell me when initialization was complete before the search state was entered. I'm interested to see if anyone else thinks this would be useful. This just makes sure that the initialization phase is complete before the receiver starts detecting packets.
```c++
/*This is the first state when the flowgraph starts. 
It is never entered into again, once it transitions the receiver to the SEARCH state.*/
case INIT: {
			int k;
			for(k = 0; k < ninput_items[2]; k++) {
				d_initSamples++;
/*the autocorrelation value must be low, and d_initSamples must have been received 
since starting the flow graph.*/
				if(in_cor[k] < d_threshold && d_initSamples > 1e7) {
					dout << "Initialization complete!" << std::endl;
					d_state = SEARCH;
					d_initSamples = 0;
					break;
				}
			}
			consume_each(k);
			return 0;
		} 
```
I was surprised that this initialization phase is captured by the autocorrelation output, which prompted me to add this code to my build. The code is very helpful. 

The values for d_threshold, d_initSamples, and the GAIN in the flow graph may need to be adjusted depending on each users set-up, perhaps d_initSamples could be a block parameter.

bastibl, 2016-01-07:
AFAIK, this is since the DC offset compensation needs some time to get a good estimate for the mean. During that time the DC component triggers frame detection due to its autocorrelation. That should not be much of a problem, but only cause some false positives in the beginning.

If you want to avoid that, you can just skip the first n samples with the Head block of GNU Radio.

bkihei, 2016-01-07:
Thanks so much for the reply. The Skip Head block worked perfectly.

bastibl, 2016-01-07:
great!

closed 2016-01-07 by bastibl