bastibl/gr-ieee802-11 #28: Compile error

nextLVL, 2016-03-03:
Hi, I have these problems when I try to compile the gr-ieee802-11. Please help me! T_T
```
[  3%] Built target ieee802_11_generated_includes
[  7%] Built target ieee802_11_generated_sources
[ 10%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/ofdm_equalize_symbols.cc.o
/home/adolfo/Documentos/OFDM/gr-ieee802-11/lib/ofdm_equalize_symbols.cc: In member function ‘virtual int ofdm_equalize_symbols_impl::general_work(int, gr_vector_int&, gr_vector_const_void_star&, gr_vector_void_star&)’:
/home/adolfo/Documentos/OFDM/gr-ieee802-11/lib/ofdm_equalize_symbols.cc:60:76: error: ‘get_tags_in_window’ was not declared in this scope
   get_tags_in_window(tags, 0, i, i + 1, pmt::string_to_symbol("ofdm_start"));
                                                                            ^
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/ofdm_equalize_symbols.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [all] Error 2
```

bastibl, 2016-03-03:
Hi, either the include directory for GNU Radio points to the wrong directory or you have a very old GNU Radio installation (check with ``gnuradio-config-info -v``)

bastibl, 2016-03-08:
I close this for now. I guess the problem is solved.

closed 2016-03-08 by bastibl

nextLVL, 2016-03-10:
Sorry for the delay , the version I have is the 3.7.2.1

bastibl, 2016-03-10:
...this version is from 2013. You should consider updating.

SamanthaJDS, 2016-07-25:
Hello,
I have version 3.7.9.2 and I am having the same issue.  Where might I find the include directory for GNU Radio (gnuradio/include?) to fix this and where should it be pointing to?

bastibl, 2016-07-26:
This particular error means that the `get_tags_in_range` function is not available. Since this function is available since 3.7.3, I guess, you have another problem or you have multiple versions installed.