bastibl/gr-ieee802-11 #29: problem installing gr-foo and gr-ieee802_11

abhignans, 2016-03-08:
im trying to install gr-ieee802-11 in gnu radio but the it shows like this
[image]

i have did the following
git clone https://github.com/bastibl/gr-foo.git
cd gr-foo
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig

here is the terminal window

git clone https://github.com/bastibl/gr-foo.git
fatal: destination path 'gr-foo' already exists and is not an empty directory.
abhi@abhignan:~$ cd gr-foo
abhi@abhignan:~/gr-foo$ mkdir build
mkdir: cannot create directory ‘build’: File exists
abhi@abhignan:~/gr-foo$ cd build
abhi@abhignan:~/gr-foo/build$ cmake ..
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
Checking for GNU Radio Module: RUNTIME
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: PMT
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
GNURADIO_PMT_FOUND = TRUE
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/gnuradio/GrTest.cmake:45 (get_target_property):
  Policy CMP0026 is not set: Disallow use of the LOCATION target property.
  Run "cmake --help-policy CMP0026" for policy details.  Use the cmake_policy
  command to set the policy and suppress this warning.

  The LOCATION property should not be read from target "test-foo".  Use the
  target name directly with add_custom_command, or use the generator
  expression $<TARGET_FILE>, as appropriate.

Call Stack (most recent call first):
  lib/CMakeLists.txt:80 (GR_ADD_TEST)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- 
-- Checking for module SWIG
-- Disabling SWIG because version check failed.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/abhi/gr-foo/build
@abhignan:~/gr-foo/build$ make
[ 63%] Built target gnuradio-foo
[ 81%] Built target test-foo
[100%] Built target pygen_python_e2406
@abhignan:~/gr-foo/build$ sudo make install
[ 63%] Built target gnuradio-foo
[ 81%] Built target test-foo
[100%] Built target pygen_python_e2406
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/include/foo/api.h
-- Up-to-date: /usr/local/include/foo/burst_tagger.h
-- Up-to-date: /usr/local/include/foo/packet_dropper.h
-- Up-to-date: /usr/local/include/foo/packet_pad.h
-- Up-to-date: /usr/local/include/foo/packet_pad2.h
-- Up-to-date: /usr/local/include/foo/periodic_msg_source.h
-- Up-to-date: /usr/local/include/foo/wireshark_connector.h
-- Up-to-date: /usr/local/include/foo/rtt_measure.h
-- Up-to-date: /usr/local/lib/libgnuradio-foo.so
-- Up-to-date: /usr/local/lib/python2.7/dist-packages/foo/__init__.py
-- Up-to-date: /usr/local/lib/python2.7/dist-packages/foo/__init__.pyc
-- Up-to-date: /usr/local/lib/python2.7/dist-packages/foo/__init__.pyo
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_burst_tagger.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_packet_dropper.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_packet_pad.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_packet_pad2.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_periodic_msg_source.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_rtt_measure.xml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/foo_wireshark_connector.xml
@abhignan:~/gr-foo/build$ sudo make ldconfig
make: *** No rule to make target 'ldconfig'.  Stop.
@abhignan:~/gr-foo/build$ sudo ldconfig
@abhignan:~/gr-foo/build$ 

even after this gnuradio gives the same error.

after gr-foo 
i have also tried 
git reset --hard 00beec517d096e4f2f6a45250750369db90dc104
 but no difference.
and tried this.

export PYTHONPATH=/opt/local/lib/python2.7/site-packages

still no effect.please let me know what to do.

Thank you

bastibl, 2016-03-08:
Hi, please start the flow graph in a terminal that might provide more information about what's going wrong.

Check that the Python module is found by starting python and type ``import ieee802_11``.

That doesn't match your install path
```
export PYTHONPATH=/opt/local/lib/python2.7/site-packages
```

You should do 
```
export PYTHONPATH=/usr/local/lib/python2.7/dist-packages
```

Hope it helps.

abhignans, 2016-03-08:
Thanks for the speedy reply.

i have exported and here it is 

abhi@abhignan:~$ export PYTHONPATH=/usr/local/lib/python2.7/dist-packages
abhi@abhignan:~$ gnuradio-companion 
linux; GNU C++ version 5.2.1 20150911; Boost_105800; UHD_003.009.001-0-unknown

<<< Welcome to GNU Radio Companion 3.7.8 >>>

Preferences file: /home/abhi/.grc
Block paths:
	/usr/local/share/gnuradio/grc/blocks
	/home/abhi/.grc_gnuradio
	/usr/share/gnuradio/grc/blocks

Loading: "/home/abhi/Desktop/untitled.grc"
>>> Done

Showing: "/home/abhi/Desktop/untitled.grc"

Loading: "/home/abhi/Desktop/sdr/gr-ieee802-11-master/examples/wifi_phy_hier.grc"
>>> Error: Connection between ieee802_11_ofdm_decode_mac_0(out) and pad_sink_2(in) could not be made.
	sink key "in" not in sink block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 338, in import_data
    raise LookupError('sink key "%s" not in sink block keys' % sink_key)
LookupError: sink key "in" not in sink block keys
>>> Error: Connection between pad_source_1(out) and ieee802_11_ofdm_mapper_0(in) could not be made.
	source key "out" not in source block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 332, in import_data
    raise LookupError('source key "%s" not in source block keys' % source_key)
LookupError: source key "out" not in source block keys
>>> Done

Showing: "/home/abhi/Desktop/sdr/gr-ieee802-11-master/examples/wifi_phy_hier.grc"

Showing: "/home/abhi/Desktop/untitled.grc"

Loading: "/home/abhi/Desktop/sdr/gr-ieee802-11-master/examples/wifi_phy_hier.grc"
>>> Error: Connection between ieee802_11_ofdm_decode_mac_0(out) and pad_sink_2(in) could not be made.
	sink key "in" not in sink block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 338, in import_data
    raise LookupError('sink key "%s" not in sink block keys' % sink_key)
LookupError: sink key "in" not in sink block keys
>>> Error: Connection between pad_source_1(out) and ieee802_11_ofdm_mapper_0(in) could not be made.
	source key "out" not in source block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 332, in import_data
    raise LookupError('source key "%s" not in source block keys' % source_key)
LookupError: source key "out" not in source block keys
>>> Done

Showing: "/home/abhi/Desktop/sdr/gr-ieee802-11-master/examples/wifi_phy_hier.grc"
^CTraceback (most recent call last):
  File "/usr/bin/gnuradio-companion", line 128, in <module>
    main()
  File "/usr/bin/gnuradio-companion", line 121, in main
    ActionHandler(args, Platform())
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/gui/ActionHandler.py", line 75, in __init__
    gtk.main()
.

still the same

Thank you

bastibl, 2016-03-08:
please connect the blocks on the top by hand, clicking on the handles. With GNU Radio updates, these connections break from time to time.
Then the connection errors should be gone.
It seems, however, that GRC cannot load the python library as the Packet Header Generator shows some error.

So please test import of the module in a terminal

abhignans, 2016-03-08:
well i get about the connections but all i am worried is the value of the block which says
[image]
[image]

 and when i try to import in the terminal
it says
python
Python 2.7.10 (default, Oct 14 2015, 16:09:02) 
[GCC 5.2.1 20151010] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> export PYTHONPATH=/usr/local/lib/python2.7/dist-packages
  File "<stdin>", line 1
    export PYTHONPATH=/usr/local/lib/python2.7/dist-packages
                    ^
SyntaxError: invalid syntax
>>> 

i even tried importing like this
abhi@abhignan:~$ export PYTHONPATH=/usr/local/lib/python2.7/dist-packages

but no good.

im confused if i am doing it right.
please guide me through this

Thank you

bastibl, 2016-03-08:
- export the pythonpath
- type ``python`` to start python
- type ``import ieee802_11`` to import the module

abhignans, 2016-03-08:
i have tried this and following error shows up 
$ python
Python 2.7.10 (default, Oct 14 2015,) 
[GCC 5.2.1 20151010] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
ImportError: No module named ieee802_11_swig
>>> 

when i have tried installing it from the git it shows a new error as
git clone git://github.com/bastibl/gr-ieee802-11.git
ERROR: ld.so: object '/usr/local/lib/libgnuradio-ieee802-11.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Cloning into 'gr-ieee802-11'...
ERROR: ld.so: object '/usr/local/lib/libgnuradio-ieee802-11.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
remote: Counting objects: 1736, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1736 (delta 0), reused 0 (delta 0), pack-reused 1732
Receiving objects: 100% (1736/1736), 655.43 KiB | 0 bytes/s, done.
Resolving deltas: 100% (1166/1166), done.
Checking connectivity... ERROR: ld.so: object '/usr/local/lib/libgnuradio-ieee802-11.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
done.

bastibl, 2016-03-08:
Somehow, your system seems to be a bit messed up. Looks like have some LD_PRELOAD set.

Since it claims that there is 
```
No module named ieee802_11_swig
```
did you check if it is there?

Were there any error during the installation of the gr-ieee802-11, or did you already have problems checking out the git repository?

abhignans, 2016-03-08:
i rebooted my system and was able to reinstall without any isuues.
but when i open the wifi_phy_hier.grc, i get the following error.

Loading: "/home/abhi/gr-ieee802-11/examples/wifi_phy_hier.grc"
>>> Error: Connection between ieee802_11_ofdm_decode_mac_0(out) and pad_sink_2(in) could not be made.
	sink key "in" not in sink block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 338, in import_data
    raise LookupError('sink key "%s" not in sink block keys' % sink_key)
LookupError: sink key "in" not in sink block keys
>>> Error: Connection between pad_source_1(out) and ieee802_11_ofdm_mapper_0(in) could not be made.
	source key "out" not in source block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 332, in import_data
    raise LookupError('source key "%s" not in source block keys' % source_key)
LookupError: source key "out" not in source block keys
>>> Done

Showing: "/home/abhi/gr-ieee802-11/examples/wifi_phy_hier.grc"

bastibl, 2016-03-08:
...this is again the same error as in the beginning, telling you that the blocks are not connected. Come on... This is really goin nowhere.

abhignans, 2016-03-08:
im sorry to bother you bastibl...my bad i didnt add the screen shot later what i want to ask.
i have connected them manually and been having problems with the headers.

it says there is no swig module but in the module i can find it.
abhi@abhignan:~/gr-ieee802-11/build$ python
Python 2.7.10 (default, Oct 14 2015, 16:09:02) 
[GCC 5.2.1 20151010] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
ImportError: No module named ieee802_11_swig
>>> 

[image]

bastibl, 2016-03-08:
...this is the source directory, which you downloaded through git. The interesting bit is whether you installed it. Should be in ``/usr/local/lib/python2.7/dist-packages/ieee802_11/`` on your PC

abhignans, 2016-03-08:
nope. i have installed it but ii dont find any swig file in the /usr/local/lib/python2.7/dist-packages/ieee802_11/.

[image]

bastibl, 2016-03-08:
Then either all installations went wrong, or you have multiple parallel installations and GNU Radio uses the broken one.

abhignans, 2016-03-08:
well i am confused..
my system is even messed up by trying all possible ways to get around.
anyways. i am going to reinstall my ubuntu and try with a fresh install.
do you think i should try that or is there any other work around.

bastibl, 2016-03-08:
don't you have a build log from installing gr-ieee802-11?

abhignans, 2016-03-08:
ye i do have the build log from installing gr-iee802-11
here is it

git clone git://github.com/bastibl/gr-ieee802-11.git
Cloning into 'gr-ieee802-11'...
remote: Counting objects: 1736, done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 1736 (delta 0), reused 0 (delta 0), pack-reused 1732
Receiving objects: 100% (1736/1736), 655.43 KiB | 0 bytes/s, done.
Resolving deltas: 100% (1166/1166), done.
Checking connectivity... done.
abhi@abhignan:~$ cd gr-ieee802-11
abhi@abhignan:~/gr-ieee802-11$ mkdir build
abhi@abhignan:~/gr-ieee802-11$ cd build
abhi@abhignan:~/gr-ieee802-11/build$ cmake ..
-- The CXX compiler identification is GNU 5.2.1
-- The C compiler identification is GNU 5.2.1
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
Checking for GNU Radio Module: RUNTIME
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.8
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_RUNTIME: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
-- checking for module 'gnuradio-digital'
--   found gnuradio-digital, version 3.7.8
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_DIGITAL: /usr/lib/x86_64-linux-gnu/libgnuradio-digital.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FFT
-- checking for module 'gnuradio-fft'
--   found gnuradio-fft, version 3.7.8
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_FFT: /usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- checking for module 'gnuradio-filter'
--   found gnuradio-filter, version 3.7.8
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_FILTER: /usr/lib/x86_64-linux-gnu/libgnuradio-filter.so;/usr/lib/x86_64-linux-gnu/libgnuradio-fft.so;/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- checking for module 'gnuradio-runtime'
--   found gnuradio-runtime, version 3.7.8
 * INCLUDES=/usr/include
 * LIBS=/usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so
-- Found GNURADIO_PMT: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so;/usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so  
GNURADIO_PMT_FOUND = TRUE
[... 60 lines cut ...]
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_decode_mac.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_decode_signal.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_equalize_symbols.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_mac.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_mapper.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_parse_mac.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_sync_long.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/ofdm_sync_short.h
-- Installing: /usr/local/include/ieee802-11/gnuradio/ieee802_11/wifi_signal_field.h
-- Installing: /usr/local/lib/libgnuradio-ieee802_11.so
-- Installing: /usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.py
-- Installing: /usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.pyc
-- Installing: /usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.pyo
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_sync_long.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ether_encap.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_chunks_to_symbols.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_moving_average_xx.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_decode_mac.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_decode_signal.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_sync_short.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_mapper.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_parse_mac.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_equalize_symbols.xml
-- Installing: /usr/local/share/gnuradio/grc/blocks/ieee802_11_ofdm_mac.xml
abhi@abhignan:~/gr-ieee802-11/build$ sudo ldconfig

bastibl, 2016-03-08:
```
- Checking for module SWIG -- Disabling SWIG because version check failed.
```

Please install another swig version.

abhignans, 2016-03-08:
finally got it to work by install different version of swig..
i am really sorry i overlooked it.
now everything is fine but except this.
it says cannot create connection even after manually connecting them.
here is the error
>>> Error: Connection between ieee802_11_ofdm_decode_mac_0(out) and pad_sink_2(in) could not be made.
	sink key "in" not in sink block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 338, in import_data
    raise LookupError('sink key "%s" not in sink block keys' % sink_key)
LookupError: sink key "in" not in sink block keys
>>> Error: Connection between pad_source_1(out) and ieee802_11_ofdm_mapper_0(in) could not be made.
	source key "out" not in source block keys
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gnuradio/grc/base/FlowGraph.py", line 332, in import_data
    raise LookupError('source key "%s" not in source block keys' % source_key)
LookupError: source key "out" not in source block keys
>>> Done

Showing: "/home/abhi/gr-ieee802-11/examples/wifi_phy_hier.grc"
>>> Error: Cannot create connection.
.

bastibl, 2016-03-09:
I never had this problem. Are you sure you did not change/store the flow graphs?

abhignans, 2016-03-09:
yes i am sure i did not change any flow graphs.
just to make sure i have downloaded the whole zip from git and opened the file after extracting it now and tried again and it says same error. and the line between them is dotted .
[image]

bastibl, 2016-03-09:
please scroll down

abhignans, 2016-03-09:
well, I was stupid.
Thank you for being so much patient and baring my stupidity.
I truly appreciate your help.

Thanks alot.

closed 2016-03-09 by abhignans