bastibl/gr-ieee802-11 #30: Problems using osmosdr_source/hackrf to RX

drtyhlpr, 2016-03-27:
Hi, thanks for this great project. I tried to use osmosdr_source instead of uhd - RX only - I see alot of hex data aka possible packets from OFDM_Decode_MAC (Debug enabled) but all of seem to have a wrong checksum:

```
Decode MAC: frame start...
received complete frame - decoding
coding rate 0.5
checksum wrong  -- dropping).
```

If I use wifi_rx I also don't see any error rate information - and therefore I think this is related to the mentioned problems. Frame Error Rate: None

- I tried different sample rates, offsets and chan_est settings and antennas
- wifi_loopback works AND shows low error rate (instead of None)
- increased shared mem and volk_profile - done

so I am a little bit out of ideas - I will go on reading source and placing debug console output about checksum calculation etc.

ps. wireshark file is empty 24bytes

is it possible to use osmosdr_source/hackrf (RX-only) at all? any other ideas / of  course I am willing to read source and produce PRs.

thanks aggain

bastibl, 2016-03-27:
Hi, depends what you are actually trying to do. Changing the sample rate, for example, doesn't make much sense if you are trying to pick up signals from WiFi card.
I would recommend you checkout the readme and try to find out where things break. (i.e. are frames detected, is the signal field decoded, etc.).

drtyhlpr, 2016-03-27:
I want to sniff 802.11 AP beacons and Client probe requests and watch them in wireshark, testing channel switching etc. yes I did read the readme but did not find mutch info about the frame processing (if you find time maybe you can add 2 or 3 words to the "RX frames from a WiFi card" section) - I am currently reading your papers and source to understand it.

bastibl, 2016-03-27:
Its really hard to help, as everything I know is that it doesn't work for you... Did you try different gain settings? Are you sure its 11a/g and not compatibility mode? Are there warnings? Are there overruns? Is there a lot of interference on the channel? ....and so on
Most blocks have log and debug output. You should try to understand where things go wrong. Are frames detected (sync short). Is the signal field decoded (decode signal field). (beacons should have the same size and MCS).
It's really hardly possible to come up with a comprehensive answer, documenting everything that could go wrong :-(

drtyhlpr, 2016-03-27:
thanks for all your ideas! - my inital question more was about: Is it possible at all to use a low cost sdr (looking at the few infos I remember about phy wifi layer, collidetection, timing etc...) I was asking myself maybe the device-hw is the limitation. I am realy reading into your papers now.

ps. I pasted the above - heavy shortend lines - because I get very fast a few MB of log data if I enable debug of OFDM_Decode_MAC - so without reading the source - or grep -r the right source files I tought that these messages show that the frame processing works - till checksuming (which of course does not say mutch about the source of the error if we talk about RF)

```
Decode MAC: frame start...len X, symbols Y, encoding [0-6]
many lines of "copy one symbol"
received complete frame - decoding
coding rate 0.5
checksum wrong  -- dropping.
```

I am happy to have this code and I realy do not expect you to solve my problems :)

bastibl, 2016-03-27:
I don't have a HackRF, but several people reported to use it successfully with this module.

Unfortunately, the log doesn't tell you much. It means that something triggered frame detection and the 1-bit parity check in the signal field was OK, so the receiver tried to decode it.
Now the question is whether this was an actual frame or noise.
Since the frame format appears to be random for you, I guess its noise.

If you see such large amounts of data it might be that you have an uncompensated DC offset, which triggers frame detection all the time. (Frame detection is based on autocorrelation, and the DC offset has a large autocorrelation :-))

I would recommend, you tune to an idle channel to check whether frame detection is still triggered (which would, of course, be bad.). Hope it helps

drtyhlpr, 2016-03-28:
I am taking some days to get indepth understanding of frame detection ... starting at chosen windowsize 48 (reducing all - and create myself a poc to simply see detected preambles) and or maybe order a usrp :)

I think that I have learned last night that the data I receive is not "just" noise (fingers-crossed) - so I can defnitly switch between busy (OFDM_decode_mac->debug->console hyperflood) and non-busy 802.11 channels (with very very few "copy one symbols", or "frame start" messages). Reducing and increasing the gain also works "as expected".

ps. I am stil not sure about the right hackrf freqcor (ppm) and "DC Offset Mode" settings (auto: "periodically find the best solution to compensate DC offset" - hmmm yes but more details please :)

bastibl, 2016-04-07:
I close this for now. Feel free to open another issue in case there are still problems.

closed 2016-04-07 by bastibl