bastibl/gr-ieee802-11 #31: some problems when installing gr-foo

GladToLovelyj, 2016-03-30:
Hi,thanks for this wonderful project! But i have some problems when i installed the gr-foo in the step"make",it shows as follows:
`Linking CXX executable foo_swig_swig_2d0df
Swig source
/home/loveyan/gr-foo/swig/foo_swig.i:3: Error: Unable to find 'gnuradio.i'
swig/CMakeFiles/foo_swig_swig_2d0df.dir/build.make:97: recipe for target 'swig/foo_swig_swig_2d0df' failed
make[2]: *** [swig/foo_swig_swig_2d0df] Error 1
CMakeFiles/Makefile2:286: recipe for target 'swig/CMakeFiles/foo_swig_swig_2d0df.dir/all' failed
make[1]: *** [swig/CMakeFiles/foo_swig_swig_2d0df.dir/all] Error 2
Makefile:126: recipe for target 'all' failed
make: *** [all] Error 2
`
I'm looking forward to your help,thank you very much!

bastibl, 2016-03-30:
Looks like you GNU Radio installation is broken. You could try installing another OOT module. I guess they will fail too.

GladToLovelyj, 2016-03-30:
Thank you very much! I have successfully installed GNU Radio 3.7.8 before this with some warnings.
Could you tell me how to solve this problem?
thanks for your reply!

bastibl, 2016-03-30:
Please see the GNU Radio Wiki:
http://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGR
If you still have problems you can ask for help on the mailing list. (If you write to the list, you should consider actually mentioning the warnings that you got.)

GladToLovelyj, 2016-03-30:
Thank you! I have reinstalled the GNU Radio3.7.8,and this problem has been solved.But now when I open wifi_phy_hier.grc,some blocks missed.It's as follows:
[image]
[image]

bastibl, 2016-03-30:
Please read the readme
https://github.com/bastibl/gr-ieee802-11#check-message-port-connections

closed 2016-04-07 by GladToLovelyj