bastibl/gr-ieee802-11 #33: ieee802_11 module in wifi_phy_hier

avinashkalyanaraman, 2016-06-09:
Hi, 

I tried opening wifi_phy_hier via GRC and I noticed it red-ed as a missing module. I set PYTHONPATH by looking at the makefile as : "export PYTHONPATH=/usr/local/lib/python2.7/site-packages/". However my GRC now crashes as: 

Loading: "/Users/avinash/usrp/gr-ieee802-11/examples/wifi_phy_hier.grc"
Fatal Python error: PyThreadState_Get: no current thread
Abort trap: 6

When I do /usr/bin/python2.7
import ieee802_11

The import is successful

However, the python which GRC is picking is /opt/local/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python

I think there's a Python mismatch. Please let me know if there's any quick way to fix this ?

Thanks!

bastibl, 2016-06-09:
you have to compile GNU Radio and OOT modules with the same python libs. Use cmake to configure PYTHON_LIBRARY and PYTHON_INCLUDE_DIR.

avinashkalyanaraman, 2016-06-09:
That fixed it - thanks!

closed 2016-06-09 by avinashkalyanaraman