bastibl/gr-ieee802-11 #34: CSI in gr-ieee802-11

avinashkalyanaraman, 2016-06-10:
Hi,

Is there a way I can get the channel frequency response from the preamble similar to how the Intel 5300 calculates it ? 

Is there a way I can modify the lib/equalizer//ls.cc or lib/equalizer/lms.cc to quickly extract the CSI for each successfully received frame ?

bastibl, 2016-06-10:
yes, checkout the development branch and look at the its LS equalizer.

avinashkalyanaraman, 2016-06-10:
Thanks.

I am trying to run wifi_loopback in the development branch. Here, I first see the grc has some missing connections.

[image]

I tried two things to fix the connections:
1. To connect the WiFi PHY Hier carrier output to a null sink and disabling the PDU to Tagged and Scope sinks

[image]

I see the following error
Using Volk machine: avx_64_mmx_orc
MAPPER: encoding: 5
set_min_output_buffer on block 18 to 397056
set_min_output_buffer on block 20 to 397056
set_min_output_buffer on block 22 to 397056
set_min_output_buffer on block 23 to 397056
set_min_output_buffer on block 25 to 397056
set_min_output_buffer on block 38 to 96000
Traceback (most recent call last):
  File "/Users/avinash/usrp/gr-ieee802-11-development/gr-ieee802-11/examples/wifi_loopback.py", line 301, in <module>
    main()
  File "/Users/avinash/usrp/gr-ieee802-11-development/gr-ieee802-11/examples/wifi_loopback.py", line 295, in main
    tb = top_block_cls()
  File "/Users/avinash/usrp/gr-ieee802-11-development/gr-ieee802-11/examples/wifi_loopback.py", line 230, in __init__
    self.connect((self.wifi_phy_hier_0, 1), (self.blocks_null_sink_0, 0))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/hier_block2.py", line 47, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/hier_block2.py", line 110, in connect
    self.primitive_connect(*args)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/runtime_swig.py", line 5331, in primitive_connect
    return _runtime_swig.top_block_sptr_primitive_connect(self, *args)
ValueError: port number 1 exceeds max of 0

2.
 Next, I tried connecting the scope directly to the WifiPhyHier block, and I see a similar output.
[image]

Using Volk machine: avx_64_mmx_orc
MAPPER: encoding: 5
set_min_output_buffer on block 30 to 397056
set_min_output_buffer on block 32 to 397056
set_min_output_buffer on block 34 to 397056
set_min_output_buffer on block 35 to 397056
set_min_output_buffer on block 37 to 397056
set_min_output_buffer on block 50 to 96000
Traceback (most recent call last):
  File "/Users/avinash/usrp/gr-ieee802-11-development/gr-ieee802-11/examples/wifi_loopback.py", line 315, in <module>
    main()
  File "/Users/avinash/usrp/gr-ieee802-11-development/gr-ieee802-11/examples/wifi_loopback.py", line 309, in main
    tb = top_block_cls()
  File "/Users/avinash/usrp/gr-ieee802-11-development/gr-ieee802-11/examples/wifi_loopback.py", line 245, in __init__
    self.connect((self.wifi_phy_hier_0, 1), (self.wxgui_scopesink2_0, 0))
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/hier_block2.py", line 47, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/hier_block2.py", line 110, in connect
    self.primitive_connect(*args)
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/gr/runtime_swig.py", line 5331, in primitive_connect
    return _runtime_swig.top_block_sptr_primitive_connect(self, *args)
ValueError: port number 1 exceeds max of 0

Could you please let me know how I might be able to fix the connections to get the development branch running ?

Thanks!

bastibl, 2016-06-10:
The PHY flow graph changed in develop. You have to open and rebuild it. Then refresh or restart GRC.

avinashkalyanaraman, 2016-06-10:
I had rebuilt the PHY flowgraph, but restarting GRC fixed it. Thanks!

I am trying to understand the estimator code in ls.cc. I notice by looking at stdout, that the value of n is monotonically increasing to 38. What does "n" here mean ? 

The 64 here represents the 64 sub-carriers ? 

Thanks!

bastibl, 2016-06-10:
n is the OFDM symbol. 64 is for the 64 subcarriers.

avinashkalyanaraman, 2016-06-13:
Thanks a few other quick questions.

I am guessing d_H[i] here for each sub-carrier contains the CSI of the sub-carrier. Why is this calculated on n=1 ?   

Could you also please let me know what the granularity of the CSI calculation is - i.e. how often is the CSI calculated and on what data ?

Thanks!

bastibl, 2016-06-13:
it's calculated after the second (n==1) symbol, i.e., after the long preamble.

The second point depends on the channel estimation algorithm. LS is the simplest one, which estimates the channel after the long preamble and keeps it constant for the rest of the frame.

See the other algorithms for more complex algorithms.

closed 2016-06-13 by bastibl