bastibl/gr-ieee802-11 #35: SDR to SDR communication - message not transmitted

avinashkalyanaraman, 2016-06-13:
Hi,

Loopback works fine for me. I am now trying to communicate between two USRP N210s with WBX.

While running Rx and Tx separately I get 'U' on the console repeatedly. I increased the shared-memory to 2GB. I modified the transmitter USRP sink and receiver USRP source to have a carrier frequency of 900MHz. I also set the Frame Equalizer to 900MHz at the receiver.

**Transmitter**
[image]

**Receiver**:
[image]

[image]

Please let me know how I might be able to fix this.

Thanks!

avinashkalyanaraman, 2016-06-13:
At the receiver, I see a series of 'D' in the console.

bastibl, 2016-06-13:
This can have many reasons, a lot of which are covered in the readme. Please go through them and, in case there are still problems, provide some more details about what you tried.

avinashkalyanaraman, 2016-06-14:
Yes, I did try the following ones.

1. I increased shared memory usage to 4GB (sudo sysctl -w kern.sysv.shmmax=4294967296)
2. I ran volk_profile & restarted GRC
3. I noticed same set of problems while bypassing the switch - i.e. connecting the USRP directly via ethernet to my computer.
4. I enabled real-time scheduling via GRC and the same problems continued.

I haven't tried the sysconf change alone. I am not sure what that is. Could that be a problem ?

The problems are a series of 'U's when transmitter alone runs. And then a series of Ds when the receiver is started. Please note that my setup is to have Tx and Rx USRPs connected to the same computer via a switch.

bastibl, 2016-06-14:
The Us and Ds indicate buffer overruns and underruns. You could try using a smaller bandwidth or a faster PC. Maybe the PCs just can't produce/consume the data fast enough.

On TX side, there isn't much that can go wrong (the frame is precomputed and then streamed to the device). If there are underruns you have to do the sysconfig changes to optimize network buffers.

On RX side, make sure that the devices is not triggered through noise (sync short log/debug option). This happens sometimes through DC offsets and might cause a huge computational overhead.

If there are still problems, try to isolate RX and TX by using a normal 11a (5GHz) transceiver to generate/receive frames.

avinashkalyanaraman, 2016-06-14:
Thanks. 

I tried the TX option. I am able to eliminate the underflows if I can lower the sampling rate to 5M. I thus set bandwidth to 2.5M, to be Nyqvist compliant.

On the RX side, lowering the sampling rate didn't help.  I enabled sync short logging/debug on the RX, and experimented without the transmitter. I see a lot of the following output.

gr::log :INFO: sync_short0 - frame start at 37563653
SHORT Frame!
SHORT copied 481
SHORT copied 14
gr::log :INFO: sync_short0 - frame start aDt 37564134
SHORT Frame!
SHORT copied 467
SHORT copied 14
Dgr::log :INFO: sync_short0 - frame start at 37564615
SHORT Frame!
SHORT copied 467
DSHORT copied 14
gr::log :INFO: sync_short0 - frame start at 37565096

Please let me know if this is something which I should expect or is this noise?

Thanks!

bastibl, 2016-06-14:
`I tried the TX option. I am able to eliminate the underflows if I can lower the sampling rate to 5M. I thus set bandwidth to 2.5M, to be Nyqvist compliant.`

With 5Msps you will get 5MHz. Maybe you just set your bandwidth wrong.

Regarding the RX, I don't know. If you don't send any frames, but frame detection is triggered all the time (in fact, so often that it even causes overruns). Then, yes, somethings going wrong. It's often a DC offset. If you have a N210 you can try to get rid of it by using an LO offset.

avinashkalyanaraman, 2016-06-14:
Thanks. 

1. Could you please let me know how I might be able to get rid of the DC offset for the N210? Is it via trial and error that I determine the LO offset ? I have tried the following things:

2. After calibrating according to Ettus website, and changing the LO offset to 0MHz (one of the three options), I see less verbose output on frame detection than 6 or 11MHz.  

Subset of the output:

gr::log :INFO: sync_short0 - frame start at 648000
SHORT Frame!
SHORT copied 4096
SHORT copied 4095
SHORT copied 4096
SHORT copied 2048
SHORT copied 4096
SHORT copied 2048
SHORT copied 4096
SHORT copied 2048
SHORT copied 4096
SHORT copied 1024
SHORT copied 512
SHORT copied 256
SHORT copied 128

After this change, I don't notice an overflow on the Rx, when there's no transmitter.

3. What's an indication of frame detection?  Does "SHORT copied" indicate frame has been detected ?
 
------------------
4. After calibrating the transmitter N210 similarly, I noticed the same problem as before (U on the transmitter and several several Ds on the receiver).

Could this be because the transmitter and receiver are connected to the same host computer ? 

This problem persists even after lowering the bandwidth/sampling rate to 2.5Mhz.

bastibl, 2016-06-15:
Yes, changing the LO offset is done via the radio buttons.
Shifting the spectrum is done on the FPGA and does not introduce any computational complexity for the PC.

Since frame detection (Short copied...) stopped after configuring a DC offset on sender and receiver, it was indeed a problem of a DC offset. Maybe you placed the devices directly besides each other.

If you have performance problems connecting them to one host, just use a second PC.
(Just to be sure, you use 2 USRPs don't you?)

bastibl, 2016-06-24:
I close this for now. Feel free to reopen if you still need some help.

closed 2016-06-24 by bastibl