bastibl/gr-ieee802-11 #36: wifi_loopback.grc issue

avinashkalyanaraman, 2016-06-15:
Hi,

I am trying to run wifi_loopback.grc. However, I notice nothing on the command line about the transmitted message and the plot is blank. Following is the output : 

Using Volk machine: sse4_a_64_orc
OFDM MAPPER: encoding: 0
set_min_output_buffer on block 31 to 96000
set_min_output_buffer on block 33 to 96000
set_min_output_buffer on block 35 to 96000
set_min_output_buffer on block 36 to 96000
set_min_output_buffer on block 39 to 96000
set_min_output_buffer on block 50 to 96000

<Blank after this>

The scope plot looks blank like this:

[image]

I am in the master branch, and wifi_phy_hier was built before executing the loopback.

Please let me know what might be the issue, and how I might be able to fix this.

bastibl, 2016-06-15:
Really strange. I've never seen this.

Could you please start the flow graph from the command line by executing the generated python script (it's in the examples folder). GRC might hide some errors.

avinashkalyanaraman, 2016-06-15:
Sure - here is the output of the generated python script. Looks identical as before. 

[image]

bastibl, 2016-06-15:
You could try to enable debug / log output of blocks to try to find out where stuff breaks. 
Maybe also a git reset --hard and rebuild the flow graphs to assert that nothing was changed from the version in the repo.
Any chances that gr-foo is not compiled against the same version of GNU Radio?

avinashkalyanaraman, 2016-06-15:
Thanks - I will try that and let you know how it goes.

The gr-foo is compiled against the same GR version.

I installed GnuRadio via apt-get as opposed to PyBombs. Any chance that could be an issue here?

bastibl, 2016-06-15:
yes, that version is most likely horribly outdated.

avinashkalyanaraman, 2016-06-17:
Version changing fixed it.

closed 2016-06-17 by avinashkalyanaraman