bastibl/gr-ieee802-11 #37: make: *** [all] Error 2 Install gr ieee802-11

abraham-gebru, 2016-06-23:
Dear Sir,

I want to install gr-ieee802_11 and UWICORE 802.11 MAC on Gnuradio 3.6.5. I tried installing on the new version of Gnuradio but was not successful since the gnuradio-core package is not available. 

I followed the followed the following steps.
cd
sudo apt-get install swig
sudo apt-get install libboost-all-dev
sudo apt-get install cmake-data
sudo apt-get install liblog4cpp5-dev
sudo apt-get install libitpp-dev
sudo apt-get install libcppunit-dev
sudo apt-get install gnuradio-dev
git clone https://github.com/sunyifan112358/gr-ieee802-11.git
cd gr-ieee802-11
mkdir build
cd build
cmake ..
make

While compiling these errors occurred. Since the output is long I only put here some portion of it. Please find attached for the complete compiling output error. 

[  3%] Building CXX object lib/CMakeFiles/gnuradio-ieee802_11.dir/ofdm_decode_mac.cc.o
In file included from /usr/local/include/gnuradio/math.h:30:0,
                 from /usr/include/c++/4.8/cmath:44,
                 from /usr/include/c++/4.8/random:38,
                 from /usr/include/c++/4.8/bits/stl_algo.h:65,
                 from /usr/include/c++/4.8/algorithm:62,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:42,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/local/include/gnuradio/gr_types.h:27,
                 from /usr/local/include/gnuradio/gr_runtime_types.h:27,
                 from /usr/local/include/gnuradio/gr_basic_block.h:27,
                 from /usr/local/include/gnuradio/gr_block.h:27,
                 from /home/abraham/gr-ieee802-11/include/gnuradio/ieee802_11/ofdm_decode_mac.h:21,
                 from /home/abraham/gr-ieee802-11/lib/ofdm_decode_mac.cc:17:
/usr/include/c++/4.8/cmath:102:11: error: ‘::acos’ has not been declared
   using ::acos;
           ^
/usr/include/c++/4.8/cmath:121:11: error: ‘::asin’ has not been declared
   using ::asin;
           ^
/usr/include/c++/4.8/cmath:140:11: error: ‘::atan’ has not been declared
   using ::atan;
           ^
/usr/include/c++/4.8/cmath:159:11: error: ‘::atan2’ has not been declared
   using ::atan2;
           ^
/usr/include/c++/4.8/cmath:180:11: error: ‘::ceil’ has not been declared
   using ::ceil;
           ^
/usr/include/c++/4.8/cmath:199:11: error: ‘::cos’ has not been declared
   using ::cos;
           ^
/usr/include/c++/4.8/cmath:218:11: error: ‘::cosh’ has not been declared
   using ::cosh;
           ^
/usr/include/c++/4.8/cmath:237:11: error: ‘::exp’ has not been declared
   using ::exp;
           ^
/usr/include/c++/4.8/cmath:256:11: error: ‘::fabs’ has not been declared
   using ::fabs;
           ^
/usr/include/c++/4.8/cmath:275:11: error: ‘::floor’ has not been declared
   using ::floor;
           ^
/usr/include/c++/4.8/cmath:294:11: error: ‘::fmod’ has not been declared
   using ::fmod;
           ^
/usr/include/c++/4.8/cmath:315:11: error: ‘::frexp’ has not been declared
[... 124 lines cut ...]
                 from /usr/include/c++/4.8/random:38,
                 from /usr/include/c++/4.8/bits/stl_algo.h:65,
                 from /usr/include/c++/4.8/algorithm:62,
                 from /usr/include/boost/smart_ptr/shared_ptr.hpp:42,
                 from /usr/include/boost/shared_ptr.hpp:17,
                 from /usr/local/include/gnuradio/gr_types.h:27,
                 from /usr/local/include/gnuradio/gr_runtime_types.h:27,
                 from /usr/local/include/gnuradio/gr_basic_block.h:27,
                 from /usr/local/include/gnuradio/gr_block.h:27,
                 from /home/abraham/gr-ieee802-11/include/gnuradio/ieee802_11/ofdm_decode_mac.h:21,
                 from /home/abraham/gr-ieee802-11/lib/ofdm_decode_mac.cc:17:
/usr/include/c++/4.8/cmath:241:3: note: constexpr float std::exp(float)
   exp(float __x)
   ^
/usr/include/c++/4.8/cmath:245:3: note: constexpr long double std::exp(long double)
   exp(long double __x)
   ^
make[2]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/ofdm_decode_mac.cc.o] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-ieee802_11.dir/all] Error 2
make: *** [all] Error 2

Please let me know how to fix this error.

Thanks a lot
[compiling gr-ieee802.11 error.docx](https://github.com/bastibl/gr-ieee802-11/files/329550/compiling.gr-ieee802.11.error.docx)

bastibl, 2016-06-23:
First sentence of the readme:
> this an IEEE 802.11 a/g/p transceiver for GNU Radio v3.7.

GNU Radio 3.6.5 is over three years old (i.e., older than this module). You might consider installing  a recent version through PyBombs.

abraham-gebru, 2016-06-23:
Hi Bastian,

Thanks for your quick reply,

The package (https://github.com/sunyifan112358/gr-ieee802-11/tree/master/uwicore_80211_MAC) requires that gnuradio-core is already installed and it is not available on the recent version (GNURadio3.7). They actually tried it on GNURadio 3.3.0 but I am not successful installing it. Rather I want to try it on GNURadio 3.6.5 since gnuradio-core package is available. I have to admit I am new to GNURadio and I really appreciate your help.

abraham-gebru, 2016-06-23:
configuring  on the latest version 3.7.9.2 gives the following result

 The CXX compiler identification is GNU 4.9.2
-- The C compiler identification is GNU 4.9.2
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Build type not specified: defaulting to release.
-- Boost version: 1.55.0
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.28") 
-- checking for module 'gruel'
--   found gruel, version 3.6.5
-- Found GRUEL: /usr/local/lib/libgruel.so  
-- checking for module 'gnuradio-core'
--   package 'gnuradio-core' not found
-- Could NOT find GNURADIO_CORE (missing:  GNURADIO_CORE_LIBRARIES GNURADIO_CORE_INCLUDE_DIRS) 
CMake Error at CMakeLists.txt:93 (message):
  GnuRadio Core required to compile IEEE802-15-4

-- Configuring incomplete, errors occurred!

bastibl, 2016-06-23:
> GnuRadio Core required to compile IEEE802-15-4
Looks like you are installing 15.4, but open a thread for the 802.11 module.

> -- found gruel, version 3.6.5
Furthermore, you still have 3.6.5 installed. Maybe you have two installations now on your system...

I will close this thread for now. Please check your system. Install one (recent) version of GNU Radio and try again. In the unlikely case that you still have problems, open a new thread for the module you are actually trying to install.

closed 2016-06-23 by bastibl