bastibl/gr-ieee802-11 #39: why you use these daughterboards

bsun003, 2016-07-28:
Hello bastibl,

I'm confused about the reason(s) you used daughterboards. As far as I know, the N210 can also work on up to 6 GHz which is compatible for wifi.

Best regards,

Bojun

bastibl, 2016-07-28:
I mixed up the devices from Ettus. The B210 does not need any daughterboards and goes up to 6GHz.

The N210 is only an FPGA board and  cannot be used without a daughterboard... (there are, however, also daughterboards for up to 6GHz).

closed 2016-07-28 by bastibl

bsun003, 2016-07-28:
Actually I visited the following website:
https://www.ettus.com/product/details/UN210-KIT

It says _A modular design allows the USRP N210 to operate from DC to 6 GHz, while an expansion port allows multiple USRP N210 series devices to be synchronized and used in a MIMO configuration_

That's why I'm still confused

bastibl, 2016-07-28:
...which lists "Recommended Daughterboards and Accessories". The N210 on its own will not receive anything. With the right daughterboard ("modular design"), its possible to receive up to 6GHz. Maybe you want to contact the customer support of Ettus Research, in case you still have questions.

bsun003, 2016-07-28:
OK. Thanks for your patient answers