bastibl/gr-ieee802-11 #47: Do You Have Some Related Docs, Manual or Papers?

bsun003, 2016-09-16:
Hi Bastibl,

I'm trying to use your code and do some modifications so that it can work on NI USRP 2920 with Vert 400 antennas. However, I'm a little bit confused about why you chose the values of parameters and variables. Is there any document, manual or paper that describe this project?

Thanks,

Bojun

bastibl, 2016-09-16:
The code is pretty SDR-agnostic and does not depend on the antenna.

The requested information is available on the website linked in the description.

www.wime-project.net

closed 2016-09-16 by bastibl

bsun003, 2016-09-16:
Hi Bastian,

Thanks for the reply. I visited the website you provided but I'm still confused about "SDR-agnostic" and not depend on the antenna. Did you run the code on the actual USRP because I noticed that you set the default center frequency to 5.8 GHz and have several frequency bands. The reason I asked you about the documents is that we only have the antennas that support the maximum frequency of 1.2 GHz.

Best regards,

Bojun

bastibl, 2016-09-17:
> Did you run the code on the actual USRP because I noticed that you set the default center frequency to 5.8 GHz and have several frequency bands. 

Sorry, but I don't get your problem.

If you want to send on another frequency, just change the frequency with the USRP blocks

bsun003, 2016-09-17:
Sorry for the confusion. Here are my questions:
- What do you mean "SDR-agnostic"?
- Why doesn't depend on the antenna? if the USRP can support the frequency up to 5.8 GHz but the antenna can't support this frequency band, still we can't receive the signal.
- Like you said "just change the frequency with the USRP blocks", I'm trying to change it but a little bit confused about why you set the local oscillator offset and why you set it to 0, 6 MHz and 11 MHz.
- Is there any paper about this project? I'm confused about the parameters that you chose in the block diagram (e.g. for the frame detection part, why you chose "delay" to 16; and why you chose two "moving average length" to 64 and 48; and some similar questions). If there are some papers that relate to this project, it would be very helpful for us to modify your code in our V2V project.

Hoping you don't mind too many questions.

Bojun

bastibl, 2016-09-18:
> What do you mean "SDR-agnostic"?
The module is not tuned for a certain SDR or daughterboard. You can use it with any SDR that supports the frequency and bandwidth that you plan to use.

> Why doesn't depend on the antenna? if the USRP can support the frequency up to 5.8 GHz but the antenna can't support this frequency band, still we can't receive the signal.

That's true. If you want to send on a certain frequency band, you have to use an appropriate antenna. But I don't see how that is specific to the module. It's just physics.

> Like you said "just change the frequency with the USRP blocks", I'm trying to change it but a little bit confused about why you set the local oscillator offset and why you set it to 0, 6 MHz and 11 MHz.
 
The oscillator **offset** is not related to the center frequency. It is used to tune the local oscillator out of the band of interest. If you use an N210, you can set it to 6MHz for a 10MHz bandwidth or 11MHz for a 20MHz signal, or just don't set it at all.

> Is there any paper about this project? I'm confused about the parameters that you chose in the block diagram (e.g. for the frame detection part, why you chose "delay" to 16; and why you chose two [...]

The website that I linked in my previous posts lists two papers at the top of the publications list. They are a good starting point. However, it's not that I just made up this PHY. You can just study the IEEE 802.11 standard to understand how the signal has to look. This should give you a good idea why the parameters are selected like this.

If you want to dig deeper into the topic, I recommend you go through GNU Radios [reading list](http://gnuradio.org/redmine/projects/gnuradio/wiki/SuggestedReading). In case you still have any questions, please read [this page](http://gnuradio.org/redmine/projects/gnuradio/wiki/ReportingErrors) and ask it on the GNU Radio mailing list.

bsun003, 2016-09-18:
Thanks a lot for this detailed info. They are very helpful.