bastibl/gr-ieee802-11 #50: LED Status of USRP

bsun003, 2016-11-10:
Hi Bastian,

When I run the code for TX, the TX LED of USRP blinks with a dim light which is a different status from the status when I run LabView code. What's more, when I run the RX code, I get solid LED on for RX.  What was LED status when you ran the code? Is it working properly with this LED status for TX?

My sys info:
USRP 2920s connected by MIMO cable
GNU Radio 3.7.10.1, UHD 3.11, Ubuntu 16.04

Best regards,

Bojun

bastibl, 2016-11-10:
The brightness of the TX LED depends on the frame rate and the payload size. (I assume you use LabView to also send WiFi frames, otherwise the comparison doesn't make much sense). So when you configure the same frame rate, frame size, and encoding scheme on LabView, it should look similar.

On RX side, it is always listening for frames, i.e., the LED is on al the time.

bsun003, 2016-11-10:
I see. I thought the LED status should be either on or off according to the Getting Start Guide. I used LabView in building up general PHY layer communication system, like Mod/Demod, frame detection, sync etc. In this case, according to your info, it doesn't make sense, unfortunately.

When I run both codes, sometimes the RX shows overflow code; sometimes the TX shows stdout of "X"s. Is this working properly? or does it depend on sysctl configuration?

bastibl, 2016-11-11:
> Receiving a correct constellation map means it's working correctly, right?
Yes.

Regarding the overflows, you can find some information here
https://www.wime-project.net/installation/

If that doesn't help, you have to use fast PC.

closed 2016-11-22 by bastibl

bsun003, 2017-01-16:
Thanks for the answering. It's been quite a long time but I was trying to build up a LabView code to test whether the LED has the same behavior. However, it seems time-consuming since our project goal is not to build a wifi framework. I recently noticed that if I lower the sample rate, the LED became much clear and blink which is similar to what I saw in my USRP labs. For the overrun, according to some answering from Ettus community, lowering the sample rate can solve the overrun problem.

bastibl, 2017-01-17:
The LED blinks because your PC is too slow to keep up with the incoming sample stream. Whenever the USRP has to drop samples, it stops receiving for a short time, which you can see from the LED.
When you lower the sample rate, this doesn't happen at all (or happens very rarely), which is why the LED looks more stable.

bsun003, 2017-01-18:
Actually, the LED behavior I mentioned is for TX. The RX LED is always on; I think this is because the receiver is always listening or capturing data. When I lower the sample rate, the TX LED will brighter than using faster sample rate. If I change the value of interval, it will slow the blink speed. If I use the lowest value of interval, the TX LED will be on.