bastibl/gr-ieee802-11 #51: Can't receive data in wifi_rx

abdelwahed, 2016-11-11:
Hi Bastian,

When trying wifi_tx and wifi_rx samples using two usrp N210 with the following configuration:
sampling rate : 200 KHz (higher sampling rates shows DDD and UUUU on thelaptops)
network: two usrps and laptops connected to same switch (address of sender/receiver is specified using the addr=ip specifier)

The receiver doesn't get data. The attached log shows the following:

- Warning: starting to receive new frame before old frame was complete
- checksum wrong -- dropping

Also, I tried  the rx_demo.sh in the "/apps" folder. Wireshark can't see data.

**What could be the reason behind this?**

attached logs of sender,  receiver and also detailed device configuration (using uhd_usrp_probe command).

[receiver_log.txt](https://github.com/bastibl/gr-ieee802-11/files/585142/receiver_log.txt)
[sender_log.txt](https://github.com/bastibl/gr-ieee802-11/files/585144/sender_log.txt)
[usrp_probe.txt](https://github.com/bastibl/gr-ieee802-11/files/585143/usrp_probe.txt)

Regards,
Abdo

bastibl, 2016-11-11:
This can have many problems and with the limited information, it's hard to tell.

I would recommend to go through the information here:
www.wime-project.net/installation

Also, use a faster PC and test with WiFi cards. Then you can see if the problem is on RX or TX side.
Apart from that, changing the sampling rate by that much, might cause problems. Your frames are 100 times longer.
Check your channel. With that setup, collisions might be very likely.

closed 2016-11-22 by bastibl

abdelwahed, 2016-11-23:
Hi Bastian,

Sorry for late reply. It took me a while to bring a PC (remove every thing) and setup a fresh copy of Linux.

Is there anything missing from this list that could be a cause of the error:

- Got a Dell PC with the following specification: Core i7 processor, 8 Gb RAM
- Installed latest version of Ubuntu 
-Installed gnuradio using ByBombs
-installed gr-ieee-80211 using ByBombs
-run wifi_phy_hier.grc [OK]
-run wifi_loopback.grc [working OK]
-connected the USRP N210 directly to the  PC using ethernet cable (without switch)
-run "uhd_find_devices" [USRP is reachable]
-run "wifi_rx.grc" using the previous configuration with sampling rate 5 MHz, channel 1 (in the UI's List) [no specific sender just tried to listen to wifi beacon packets) [NO OVERRUN : D letter wasn't displayed] 
-working OK but all received frames dropped due to wrong checksum 
-experimented with different channels and gain values but no progress
-daughterboard XCVR2450 couldn't be calibrated (according to the calibration utility output)

Something else could be tried?

Regards,
Abdo

bastibl, 2016-11-23:
> run "wifi_rx.grc" using the previous configuration with sampling rate 5 MHz, channel 1 (in the UI's List) [no specific sender just tried to listen to wifi beacon packets) [NO OVERRUN : D letter wasn't displayed]

Means you generated the frames yourself with another SDR? I'm asking because normal WiFi routers use 20MHz.