bastibl/gr-ieee802-11 #53: Can't ping from one pc to another...

raghu25, 2016-11-18:
I have the wifi_transceiver.grc running on both systems which are connected to usrps but when I configure the tap0 ip address to 192.168.200.1 on host1 and 192.168.200.2 on host2 I am unable to ping from one system which is connected to B200 mini usrp to another system which is connected to another B200 mini usrp. I also followed the nic.sh script but it didn't work.

It would be great to get some help 

Thank you

Hrishekessa raghu

bastibl, 2016-11-19:
This is already a rather complex setting where many things can go wrong. You will have to find out where it breaks. When you ping:

- is there an IP route to the other PC
- is there a static ARP entry for the other PC
- if so the packet should be put in TUN/TAP and reach GNU Radio. Does that work?
- does GNU Radio modulate the packet and send it out without underruns
- is the other SDR receiving the frame
- does the MAC address match so that the packet is accepted for that host
- does the IP address match, so that the packet is put into the IP stack, where it finally causes a ping reply.

This reply, then, has to pass the very same chain in reverse. Use the logging options of my module and wireshark to find out where things break.

raghu25, 2016-11-21:
Hi Basitan

Does the BSS MAC  mean Broadcast MAC or Something else?? Do I have to change the BSS MAC to the MAC address of an intermediate router/USRP along with the two host machines connected to two USRPs?
Looking forward for your answer
Thank you 
Hrishekessa Raghu

bastibl, 2016-11-21:
There is no intermediate router in this case, so the BSS is ignored. If that doesn't fit your use case, you can change that in the block that translates between WiFi and Ethernet.

closed 2016-11-27 by bastibl

raghu25, 2016-12-01:
I am able to ping from host1 to host2 after going through your steps and making the required configuration.

 I made host1 with usrp as a router with tap0 as virtual interface and I tried to access host1 with usrp by configuring the host 1's tap0 as a gateway from host2 we are able to ping google.com from terminal but we are unable to reach the www.google.com from web browser like chrome or mozilla. Any suggestion is greatly appreciated.

following are commands used on host1 to make it as a router.
sudo iptables -t nat -A POSTROUTING -o eno1 -j MASQUERADE
sudo iptables -A FORWARD -i tap0 -o eno1 -m state --state RELATED,ESTABLISHED -j ACCEPT
sudo iptables -A FORWARD -i tap0 -o eno1 -j ACCEPT

Thank you
Hrishekessa Raghu Eeti

bastibl, 2016-12-02:
I'm not sure I understand your setup, but you could try something like:
```
echo 1 > /proc/sys/net/ipv4/ip_forward
iptables -t nat -A POSTROUTING -o eno1 -j MASQUERADE
```
The others are not required.

Also check that the router has a default gateway set and that DNS is working (i.e., see if you can connect to any website through its IP)

akshayy29, 2019-07-12:
We are running 2 USRP N210 from 2 different hosts, we are trying to ping each other with the tap0 interface IP, but we are not getting successful pings.
Finally, we want to transfer text files between the 2 devices.

Can you please help us with that?

bastibl, 2019-07-15:
Before you start with the more complex scenarios, did you actually make sure that you can send or receive anything (from a normal AP and another SDR)?

akshayy29, 2019-07-15:
Like sending between 2 SDR's? What does AP exactly mean? Access point?

bastibl, 2019-07-16:
Like receiving something from another WLAN card. Or sending something to another WLAN card. Or sending something from one SDR to the other. In general, simpler setups. Immediately starting with the most complex setup is very hard to debug.

dattajigosavi, 2019-07-17:
I am having the same issue. Able to send and receive file data using general setup from one usrp device to other. But not able to send or receive to other device  using ieee 802.11.

bastibl, 2019-07-18:
setting up networking is not totally trivial. You have to configure ARP and IP. Please see the shell script in the apps folder to get an idea how that works. Other than that use Wireshark to debug, i.e., find out where it is breaking or where you don't get a response.

dattajigosavi, 2019-07-18:
It says network unreachable , how are we supposed to add ARP entries when networks are not connected.

bastibl, 2019-07-18:
You'll have to add ARP entries manually. Please see the shell script in the apps folder.

dattajigosavi, 2019-07-19:
Thank you. We are able to send and receive message using message strobe. But in between the messages there is no definite interval , and we stop receiving frame after some time. We are getting overflows(that are the D's observed (we assume that is overflow)). 
We tried to send message simultaneously , but we are able to only send or receive message at the same time , both does not take place .