bastibl/gr-ieee802-11 #55: thread[thread-per-block[18]: <block mac (26)>]: Frame too large (> 1500)

chakpongchung, 2016-11-30:
when I try to use the wifi_tx.grc to transmit a message, it says the above error.

what is that?

thread[thread-per-block[18]: <block mac (26)>]: Frame too large (> 1500)
[image]

bastibl, 2016-11-30:
It means the frame is too large (seemingly larger then 1500 byte).
I just tried and the error doesn't occur here. Maybe you want to reset the flow graph to the original version.
If this is the exact configuration that you tried, then it looks like you would receive very large frames over the UDP Server.

chakpongchung, 2016-11-30:
Thanks!

Looks like I need to fine tune the size of the message. I was using

pmt.intern("".join("hello!" for i in range(pdu_length)))

Now I changed to 

pmt.intern("hello!")

Problem resolved.

closed 2016-11-30 by chakpongchung