bastibl/gr-ieee802-11 #56: pass the message as argument to the python script or C code

chakpongchung, 2016-12-01:
Hi,

Thank you for your amazing work. I am in the middle of developing a protocol.
 Are there anyway I can pass the message as argument to the python script or C code? I also want to control the output of the terminal.

bastibl, 2016-12-01:
The easiest way is to
- send UDP packet to the Socket PDU block 
  see: https://github.com/bastibl/gr-ieee802-15-4/blob/next/utils/udp.py
- or post PMT messages to the MAC block (```post()``` function)
  see: http://gnuradio.org/doc/doxygen/classgr_1_1msg__accepter.html#a03b0afcc820d3454d29f0ada3b89e2fc

chakpongchung, 2016-12-01:
Thanks! But what I am looking for is some example code that can compile and run.

For example, in c/c++ or python, I want to have a send/receive pair of codes for ODFM wifi.
Something similar to this:

http://gnuradio.org/redmine/projects/gnuradio/repository/revisions/71127c2d4082119ea086f9f881b349922ed8cf24/entry/gr-fcd/examples/c++/fcd_nfm_rx.cc

bastibl, 2016-12-01:
the first script is ready to run. For the second approach, I'm afraid, you will have to add that line of code.

closed 2016-12-01 by bastibl