bastibl/gr-ieee802-11 #59: gr-ieee802-11 outer coder

Binetasarr, 2016-12-13:
Hi,
Hi, I've been setting up gr-ieee802-11 on a Linux with gnuradio 3.7.10.1.
I have added an outer coder in gr-ieee802.11 with gr_modtool add and make the xml file with gr_modtool makexml.
the block is available in gr-ieee802-11.

Right now I'm just trying to build "/examples/wifi_phy_hier.grc" as detailed on the main Github page. However I get the following errors in Gnuradio-companion:

Error 0:
Block - header_formatter - Variable(variable):
  Param - Value(value):
    Value "ieee802_11.signal_field()" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 1:
Block - chan_est - Parameter(parameter):
  Param - Value(value):
    Value "ieee802_11.LS" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 2:
Block - digital_packet_headergenerator_bb_0 - Packet Header Generator(digital_packet_headergenerator_bb):
  Param - Formatter Object(header_formatter):
    Value "header_formatter.formatter()" cannot be evaluated:
    name 'header_formatter' is not defined

Error 3:
Block - encoding - Parameter(parameter):
  Param - Value(value):
    Value "ieee802_11.BPSK_1_2" cannot be evaluated:
    name 'ieee802_11' is not defined

Error 4:
Block - ieee802_11_frame_equalizer_0 - WiFi Frame Equalizer(ieee802_11_frame_equalizer):
  Param - Algorithm(algo):
    Value "chan_est" cannot be evaluated:
    name 'chan_est' is not defined

Error 5:
Block - ieee802_11_mapper_0 - WiFi Mapper(ieee802_11_mapper):
  Check "$encoding in range(8)" did not evaluate.

Error 6:
Block - ieee802_11_mapper_0 - WiFi Mapper(ieee802_11_mapper):
  Param - Encoding(encoding):
    Value "encoding" cannot be evaluated:
    name 'encoding' is not defined

I'm guessing there is something I did wrong in setting everything up, but I can't figure out what. Any ideas?

bastibl, 2016-12-13:
It looks as if your PYTHONPATH is not setup correctly, as the module cannot be found.
If the path is OK, it might fail loading some libraries, but the error is hidden by GRC.
Try loading the module from the terminal to see what's going wrong.

Binetasarr, 2016-12-13:
when i try to load the wifi_hier_phy.py, i have these errors messages

Traceback (most recent call last):
  File "wifi_phy_hier.py", line 14, in <module>
    import ieee802_11
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 28, in <module>
    _ieee802_11_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_ieee802_11_swig', fp, pathname, description)
ImportError: /usr/local/lib/libgnuradio-ieee802_11.so: undefined symbol: _ZN2gr10ieee802_1111rank_metricD1Ev

Thanks for your help

bastibl, 2016-12-13:
That is a linker problem with your extensions.

Binetasarr, 2016-12-13:
Can you explain more please?
im new in gnuradio

bastibl, 2016-12-13:
I'm afraid I can't helm much with this. This is not really a problem of GNU Radio or my module.
All I know is you implemented something, but didn't link correctly.
I guess you should have a look at your cmake files.

Binetasarr, 2016-12-14:
Thank you for your help.

closed 2016-12-14 by bastibl