bastibl/gr-ieee802-11 #61: Using the graph with HackRf

ayserafettin, 2017-01-11:
Hello,
did anyone used the transceiver with HackRf as signal source?I am trying to capture wifi signals with Hack Rf and wanted to use this but  I could not calibrate it with 5.24 GHz wifi frequency. It overloads and cannot decode anything.Any help will be appreciated. Thanks

Regards
Seref

bastibl, 2017-01-11:
Yes, I know from people who operated the transceiver with the HackRF. 
Please see here for tips to troubleshoot the installation: https://www.wime-project.net/installation/
Anyhow, you will need some computational power for 20MHz WiFi. A dated laptop running a Linux in a VM, for example, might not be enough.
If you still have problems, it would be helpful to provide a more detailed description of what you did and what you mean with "it overloads".

ayserafettin, 2017-01-18:
There is not no error with hackRf but it overruns and i dont know why. It gets sometimes Beacon  frames but there are Overruns (OOOOO...). You mentioned to run volk_profile should i run it separately? and after i run what should I do wit that?

bastibl, 2017-01-18:
nothing, you just have to run it so that VOLK finds the best implementations for SIMD instructions for your particular CPU.
If frame detection is working (no overruns if there are no frames), you will have to use a faster PC.
20 MHz WiFi frames results in 20.000.000 complex floating point numbers per second. You will need an average PC to process them in real-time. (a dated laptop that is running GNU Radio in a VM, for example, will not do)

ayserafettin, 2017-01-18:
Ok, i will do that again and  many thanks for your quick reply. Best Regards

closed 2017-01-25 by bastibl