bastibl/gr-ieee802-11 #62: CSI

ZhijingLi, 2017-01-24:
Hi,

I have a problem understanding the CSI extraction. It seems to me that the input of the frame decoding is the 48 data subcarrier, which is calculated in ls.cc as vector c. The d_H contains the 64 subcarrier. So d_H should contain the data subcarrier c. But why in ls.cc for the data subcarrier in d_H is different from c?

Thanks in advance.

bastibl, 2017-01-24:
d_H is the channel. What you decode is data_subcarriers/d_H.
I don't understand your questions, but maybe that points you into the right direction.

ZhijingLi, 2017-01-24:
I see. So in ls.cc, gr_complex *in represents the CSI right?

If I want to output the CSI for the successfully decoded packets, I just need to output that 64 subcarriers in frame_equalizer_impl.cc when d_current_symbol == 2 (signal field). Am I right?

Thanks.

bastibl, 2017-01-24:
No, the Channel State Information (CSI) is `d_H`, i.e., the channel...
`in` are the subcarriers.
For the LS algorithm, `d_H` is not updated during frame reception. So you can copy `d_H` from any symbol >= 1.

closed 2017-01-24 by bastibl

ZhijingLi, 2017-01-24:
Thanks a lot!

Zhijing1128, 2017-01-26:
Hi, I still have a problem towards the CSI calculation.

I output the d_H when symbol == 1. However all of them are inf. And I saw in the code that:
for(int i = 0; i < 64; i++) {
			d_H[i] += in[i];
			d_H[i] /= LONG[i] * gr_complex(2, 0);			
		}
"d_H[i] /= LONG[i] * gr_complex(2, 0);" will make it inf, is that correct? Why it is calculated in this way?

Thanks in advance.

bastibl, 2017-01-26:
I assume you print the values wrong, or you printed the values for a degenerated signal (all zeros, for example).
The algorithms is just the normal LS algorithm. What exactly is it that you don't understand?

Zhijing1128, 2017-01-26:
Because for the first and last few long[i], they are all 0. d_H[i] /= LONG[i] * gr_complex(2, 0) will make it inf. I guess I should make them 0 or drop them when using CSI?

For the other CSI I got, which is not like the one I got from intel 5300. The absolute values are much smaller even when SNR is high, for example 0.001695 + 0.000629j for one subcarrier. But for intel 5300, it could like 10+10j. Maybe the calculation is different or I got something wrong?

Thanks.

bastibl, 2017-01-26:
It doesn't make sense to look at the guard subcarriers. The transmitter doesn't send anything on those frequencies. Therefore the receiver cannot estimate the channel for that frequencies... (you could guess based on the other carriers, but you cannot measure without a test signal)..

Apart from that it doesn't make sense to compare the absolute values since the devices are not calibrated. If you change the rx gain of the SDR by only 6dB, you will double the amplitude of the signal and also double the CSI. They are not normalized...

Zhijing1128, 2017-01-26:
It seems that the CSI from the wifi card is a relative value.

Thank you!