bastibl/gr-ieee802-11 #65: bad results

ingeda, 2017-02-03:
I'm running the project in Gnu Radio, and the loopback worksfine for every modulation scheme. But when I do the case of a Tx and Rx (I'm using 2 laptops, one Rx and the other Tx and 2 USRP connected to each one of the laptops) I get bad results. 
I made sure the USRPs are connected to each laptop and pinged them, so they're sending packets but the points in the constallation scheme are messed up, even when I change gain or any other parameter I still have bad results. 
Do we need to change the daughterboard? 
I'm using OS ubuntu 16.04 LTS . HW USRP N210, Gnu Radio 3.7.9, BW 10 Mhz and freq 5.89 Ghz (same parameters as in the project )

Thank you in advance

bastibl, 2017-02-03:
Maybe you can find something useful here:
https://www.wime-project.net/installation/

In addition to that, I would recommend to
- make sure that there are no Overruns (i.e., that the receiver can catch up with the same stream)
- check the constellation plot (it should give you a good idea if decoding works)
- try different LO offsets to tune the oscillator out of the band of interest
- make sure that frame detection works (i.e., that decoding is only triggered if a frame is sent)

ingeda, 2017-02-06:
Hello @bastibl  Bastian, 

I get this error while I'm trying to transmitt from TX to RX or even when I'm using the Transceiver:UHD Warning:
    Unable to set the thread priority. Performance may be negatively affected.
    Please see the general application notes in the manual for instructions.
    EnvironmentError: OSError: error in pthread_setschedparam

I installed VOLK, checked the daughterbouard, changed the permission file /etc/security/limits.conf by adding at the end @GROUP -rtprio 99  and still no good performance 

Any help or ideas? 

Thanks again,

bastibl, 2017-02-06:
Did you actually check the above points?
If so, could you please describe more verbose what's going wrong?

closed 2017-02-22 by bastibl