bastibl/gr-ieee802-11 #66: gr-ieee802-11 on USRP E#!@

rvincentzio, 2017-02-21:
Any idea if this will work on the new USRP embedded radio, the E312? I talk to the E312 over LAN. My computer runs Ubuntu 16.04 LTS on a VirtualBox.
I use GRC 3.7.9.
Regards,
Vincent

bastibl, 2017-02-21:
This came up several times on the GNU Radio mailing list. It doesn't work. AFAIK, the LAN functionality should be regarded as a debugging feature. It offers only very limited bandwidth.

closed 2017-02-21 by bastibl

rvincentzio, 2017-02-22:
Thanks, Bastian. Maybe I did not make it clear. I would like to run the gr-ieee802-11 program(s) on the E312 processor, as Python files. So, there is not too much to exchange with the host computer. I SSH into the E312, install and run apps and transfer files.
Regards,
Vincent

bastibl, 2017-02-22:
Sorry, I was irritated because you added so much information about the host that ssh's into the embedded device. The E312 has an ARM Cortex A9 866 MHz dual-core processor. This will not be enough to process a 10MHz or 20MHz WiFi signal. AFAIK, the E312 is more tuned towards FPGA implementations.

rvincentzio, 2017-02-22:
Thanks Bastian. I want to demonstrate the RFtap functionality on my embedded radios. I guess the RDS decoder is more appropriate.