bastibl/gr-ieee802-11 #67: Cannot use the wifi_tx on N210.

mingleiyou1988, 2017-02-22:
Hi,

I am trying to implement the 802-11 on to the N210, but wifi_tx.grc does not work with N210. Hope you can give me some suggestions.

I have tried many things to locate the problem, here is the summary of what is known:
1. If run wifi_tx and wifi_rx with two B200, everything works fine.
2. If run wifi_tx  with B200 and wifi_rx with N210, everything works fine.
3. If run wifi_tx with N210 and wifi_rx with B200, nothing receives. The transmitting indicator LED A in the front panel is not on. But a very very very dim blinking for LED A can be seen if look very very close to it. 
4. If for N210, run uhd_siggen_gui -a TX/RX -s 10M -f 1300M -x 3000 --sine, while for B2100 run uhd_fft -a type=usrp2, -f 1300M, -s 10M, everything works fine. LED A on N210 is bright on. Function test all right.

Could you give me some suggestion on this issue?

Some background infomation that might be useful:
System: 
Ubuntu 16.04 LTS
Memory: 7.5GB
Processor: Intel® Core™ i5-6300U CPU @ 2.40GHz × 4 
Graphics: Intel® HD Graphics 520 (Skylake GT2) 
OS Type: 64-bit 
Disk 62.9GB

Software:
1. gnuradio and uhd via the build-gnuradio script given in http://gnuradio.org/redmine/projects/gnuradio/wiki/InstallingGRFromSource
2. The versions are:
 gnuradio:  v3.7.10.1-211-gb49da168 
 uhd: UHD_3.11.0.git-88-g180d5f3a
 uhd on N210 :  i downloaded and burned the latest version : uhd-images_003.010.001.001-7-g63fcfb95 
3. gr-foo and gr-ieee802-11 are installed following https://www.wime-project.net/installation/.

Hardware:  two N210 with CBX Board Rev. 3 ; two B200; two laptops

If you need any more information, please let me know.

bastibl, 2017-02-22:
I would try adding some more padding in front of the frame and set different LO offsets (for the N210).

mingleiyou1988, 2017-02-23:
Problem solved.  Many many thanks for your timely reply and kind help!!

After changing the padding in front of the frame from 100 to 1000, and LO of N210 from 0 to 11M, wifi_tx for N210 and wifi_rx for B200 is working fine. Test on N210 pairs is also a pass.  

A further question:
When running the wifi_rx with N210, if there is a wifi_tx, it works fine. But occasionally there is 'D' packages indicating overflow has happened. Also if trying to receive the beacons that commercial wifi card broadcasts with samp_rate=20Mps, 'D' can be more frequently and occasionally screen is soon overwhelmed with 'D'. I have made the following change

In /etc/sysctl.conf:
net.core.rmem_max = 50000000
net.core.wmem_max = 50000000
kernel.shmmax = 2147483648

In gedit /etc/security/limits.conf
@usrp  - rtprio 99
@scbg46  - rtprio 99

I also tried to kill the irq_balance following 
https://deltafabri.wordpress.com/2016/05/25/d-packets-using-usrp2/

But the 'D' problem is still there. For communication between USRP devices, it seems D is generally harmless. Yet for the case of receiving beacons from commercial wifi cards, the MAC frame sometimes cannot be decoded correctly, which might relate to the D problem.

If run with B200, the 'O' overflow problem happens but not this serious. Actually when I test the B200, even with samp_rate=20Mps the 'O' problem is still rare cases. 

Since all tests are run on the same laptop, it has to be some configuration of the software environment.

Would you please provide some suggestions? The hardware and software setup is listed in the original post.

Many thanks for your help again!!

bastibl, 2017-02-23:
Make sure to run `volk_profile` to assert that you use the fastest SIMD instructions for your CPU.

If your PC is just too slow, there is not much you can do. But sometime the overflows are caused by oscillators leaking into the receiver (a signal with high autocorrelation triggers frame detection). That means the receiver will try to decode noise and completely overload.

Another problem that I've seen:
- the device overflows
- the device restarts
- until the device stabilizes (DC offset correction?) it produces a transient with high autocorrelation
- this signal triggers frame detection on noise
- this overloads the receiver
- goto 1

closed 2017-02-23 by mingleiyou1988

LilyZL, 2017-03-04:
Hi,

I also have the problem of the wifi_tx.grc. I'm not sure whether it does transmit something or not. The TX LED is on. But at the RX side, the PSD shows no change and no packet is received. I changed the lo offset & padding amount, but no use.

LilyZL, 2017-03-04:
The packet can be received by normal WiFi nic, right?

LilyZL, 2017-03-04:
I also tried the transceiver. Still no transmission, the constellation didn't show any change. Any idea?

bastibl, 2017-03-04:
Can you please describe in much more detail what you did and what happened.
Also please make sure that you followed the installation instructions:
https://www.wime-project.net/installation/
When setting the sample rate to 20MHz you can receive the frames with a normal WiFi card. (Make sure the card is in monitor mode.)

LilyZL, 2017-03-04:
Thanks for the reply.

For the wifi_tx.grc, I add a QT GUI frequency sink after the packet pad2 block to see whether there's signal and the question is yes. However, for the USRP sink I'm not sure whether there's transmission, the PSD on the other USRP shows no change. But if I try uhd_siggen_gui to transmit, I can get received signal on the other USRP.

LilyZL, 2017-03-05:
OMG finally I received the packets. 

But why the protocol is LLC?

bastibl, 2017-03-05:
That's according to the standard. It is a convergence protocol between the Ethernet/WiFi MAC and the layers above (like IP).