bastibl/gr-ieee802-11 #69: wifi_rx.grc with HackRF "No devices found"

muhantu, 2017-03-24:
I'm new with Linux and SDR, I want to see how is WiFi signal form, and than I try your wifi_rx.grc with my HackRF but when I run the wifi_rx.grc, this is the results
```
> Showing: "/home/muhan/sdr/gr-ieee802-11/examples/wifi_rx.grc"
> 
> Generating: '/home/muhan/sdr/gr-ieee802-11/examples/wifi_rx.py'
> 
> Executing: /usr/bin/python -u /home/muhan/sdr/gr-ieee802-11/examples/wifi_rx.py
> 
> linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown
> 
> Traceback (most recent call last):
>   File "/home/muhan/sdr/gr-ieee802-11/examples/wifi_rx.py", line 364, in <module>
>     main()
>   File "/home/muhan/sdr/gr-ieee802-11/examples/wifi_rx.py", line 352, in main
>     tb = top_block_cls()
>   File "/home/muhan/sdr/gr-ieee802-11/examples/wifi_rx.py", line 141, in __init__
>     channels=range(1),
>   File "/usr/lib/python2.7/dist-packages/gnuradio/uhd/__init__.py", line 122, in constructor_interceptor
>     return old_constructor(*args)
>   File "/usr/lib/python2.7/dist-packages/gnuradio/uhd/uhd_swig.py", line 2671, in make
>     return _uhd_swig.usrp_source_make(*args)
> RuntimeError: LookupError: KeyError: No devices found for ----->
> Empty Device Address
> 
> >>> Done
```
OS = Ubuntu 16.04 LTS
Hardware = HackRF
GNU Radio version = 3.7.9
`$ hackrf_info`
```
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c834771ccf
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00584756
```
I think my HackRF not found, but I don't know why..

bastibl, 2017-03-24:
The UHD source and sink blocks don't work with the HackRF. You will have to replace them with the sources and sinks from gr-osmosdr. You probably have them already installed. They are required to interfacing the HackRF from GNU Radio.

muhantu, 2017-03-24:
Thank you.. 
I just change the UHD source, to osmocom Source.

closed 2017-03-24 by muhantu

hugoyepezcrow1, 2019-05-21:
Dear Bastian,

Congratulations for your great program, I am student and just start in GNU-RADIO and i would like to research about attacks in physic layer in wifi networks, in my university only we have a Hackrf one. I am trying to compile gr-foo but in 8% presented errors. Maybe you can help me with a version that work with hack rf one.

Another question, maybe Dr. can you give me a recommendation about a specific subject in research in attack wifi networks.

Thank you for your help.

Regards,
Hugo

bastibl, 2019-05-21:
I don't know what happened at 8%, but it is in no way related to the HackRF. I assume it's one of the issues that got already addressed. Please see the (closed) issues of the project (i.e. gr-foo).

hugoyepezcrow1, 2019-05-21:
Thank you Bastian, I am going to search in gr-foo project, and any question I`ll tell you.

Regards

Hugo