bastibl/gr-ieee802-11 #70: Packet-level TX power change

Zhijing1128, 2017-03-29:
Hi,

how can I randomly change the TX power for every packet?

Thanks.

bastibl, 2017-03-29:
The question doesn't have much to do with the WiFi transceiver. In the end you will have to attach a tag to the start of the frame. Then, the tag has to be interpreted by the SDR sink. If you use an SDR from Ettus, this page might be relevant
http://gnuradio.org/doc/doxygen/page_uhd.html

Zhijing1128, 2017-03-30:
Thanks for the reply.

I'm thinking of changing the gain along w/ the time since we send packets as time changes. It's easier. But I'm not sure whether there will be some issues in this approach. 

Thanks in advance!

bastibl, 2017-03-30:
Sorry, but I have no idea what you are asking for. But since it's not specific to WiFi (but to GNU Radio and your SDR) and since it's not an issue (but a general question), you should post this question to the GNU Radio mailing list. Also consider describing you question in more than one sentence.

closed 2017-03-30 by bastibl