bastibl/gr-ieee802-11 #71: Receiver issue

ayserafettin, 2017-04-01:
Dear all,

I am using  gr-ieee802_11 Transceiver to analyze wifi signals. 
I receive samples and retransmit the same IQ samples wit Tx and save 
them in file, but when I try to use that IQ samples from file to receive 
them from file source some packets are not received, even though they 
are successfully transmitted and in file. Can anyone please help me what 
could be the problem? what can I do.

Best Regards

Seref

bastibl, 2017-04-01:
This can have lots and lots of reasons. So some more hints on your setup and what you already did would be very useful. Some examples:
- you have overruns while recording (your disk is not fast enough)
- you set the gain too high and experience distortions
- there was interference from other devices
- to less gain at the receiver
- DC offset and LO leakage cause false detections
and maybe 100 more possible explanations. You can find some more information at
https://www.wime-project.net/installation/
But I'm afraid it requires some more debugging on your side. Since this is not really an "issue", the topic might be better suited for the GNU Radio mailing list.

ayserafettin, 2017-04-01:
Hi,

thanks for your quick reply first of all. The title of question might not be correct. By receiving Hackrf the above mentioned things are understandable in my side but what I don not understand is that I recreated all the packets that I already received with Hackrf and try to send from file source to the receiver. and It receiver packets and sometimes in between some packets and said to be short, and so it does decode although the previous packet has the same length. and also the tx gain in my case is the same for all packets which is a constant multiplier. Can this be because of having the receiver source as file?When I send a single packet there is no problem but when send all my packets in a constant rate like once a second (I tried  sending in different time intervals ) etc.( I did this by connecting my transmitters output directly to the WIFI_HIER block as sample input and receive the mac output and also saved tx output in file and use that file in rx ) It receive for example nicely until 27th sometime it does not receive it and continue with the next. I also played with the constant  which I multiply for gain but it did not solve it. 

Anyway but many thanks for your advice.

Regards

bastibl, 2017-04-01:
Sorry I don't understand. Could you please be so kind and post your question to the GNU Radio mailing list. I have the impression that your problem is more with SDR in general than with this module.

closed 2017-04-01 by bastibl