bastibl/gr-ieee802-11 #74: ImportError: No module named _ieee802_11_swig

cit, 2017-05-08:
Hey,

I tried to install `gr-foo` and `ieee802_11` but run into problems. The compilation and installation process of both modules were without a problem. However, if I import one of the python modules I get the following error:

```
Python 2.7.13 (default, Feb 11 2017, 12:22:40) 
[GCC 6.3.1 20170109] on linux2
Type "help", "copyright", "credits" or "license" for more information.
dlopen("/usr/lib/python2.7/lib-dynload/readline.so", 2);
import readline # dynamically loaded from /usr/lib/python2.7/lib-dynload/readline.so
>>> import ieee802_11
import ieee802_11 # directory /usr/local/lib/python2.7/site-packages/ieee802_11
# /usr/local/lib/python2.7/site-packages/ieee802_11/__init__.pyc matches /usr/local/lib/python2.7/site-packages/ieee802_11/__init__.py
import ieee802_11 # precompiled from /usr/local/lib/python2.7/site-packages/ieee802_11/__init__.pyc
# /usr/lib/python2.7/plat-linux2/DLFCN.pyc matches /usr/lib/python2.7/plat-linux2/DLFCN.py
import DLFCN # precompiled from /usr/lib/python2.7/plat-linux2/DLFCN.pyc
# /usr/local/lib/python2.7/site-packages/ieee802_11/ieee802_11_swig.pyc matches /usr/local/lib/python2.7/site-packages/ieee802_11/ieee802_11_swig.py
import ieee802_11.ieee802_11_swig # precompiled from /usr/local/lib/python2.7/site-packages/ieee802_11/ieee802_11_swig.pyc
import importlib # directory /usr/lib/python2.7/importlib
# /usr/lib/python2.7/importlib/__init__.pyc matches /usr/lib/python2.7/importlib/__init__.py
import importlib # precompiled from /usr/lib/python2.7/importlib/__init__.pyc
dlopen("/usr/local/lib/python2.7/site-packages/ieee802_11/_ieee802_11_swig.so", 102);
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/site-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
  File "/usr/local/lib/python2.7/site-packages/ieee802_11/ieee802_11_swig.py", line 17, in <module>
    _ieee802_11_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/site-packages/ieee802_11/ieee802_11_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_ieee802_11_swig')
  File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
ImportError: No module named _ieee802_11_swig
```

The following files are in the site-package directory:

```
 cit@voyager > /usr/local/lib/python2.7/site-packages/ieee802_11 > tree
.
├── ieee802_11_swig.py
├── ieee802_11_swig.pyc
├── ieee802_11_swig.pyo
├── _ieee802_11_swig.so
├── __init__.py
├── __init__.pyc
└── __init__.pyo

0 directories, 7 files
```

```
cit@voyager > ~ > echo $PYTHONPATH
/usr/local/lib/python2.7/site-packages
```

Any ideas?

OS: Archlinux 4.10.13-1
SWIG Version 3.0.12
GNU Radio Companion 3.7.11

bastibl, 2017-05-08:
This came up several times. It's probably an issue with shared libraries (setup `$LD_LIBRARY_PATH`, run `sudo ldconfig`.)
Swig often hides these errors with `except ImportError` blocks.

To see the error try deleting the except block from

```
    def swig_import_helper():
        import importlib
        pkg = __name__.rpartition('.')[0]
        mname = '.'.join((pkg, '_foo_swig')).lstrip('.')
        try:
            return importlib.import_module(mname)
        except ImportError:
            return importlib.import_module('_foo_swig')
    _foo_swig = swig_import_helper()
    del swig_import_helper
```
in `foo_swig.py`, for example, and import the module in an interactive Python session `import foo`.

cit, 2017-05-08:
Your are right! After I removed the except block, I received the following error message:

```
ImportError: libgnuradio-ieee802_11.so: cannot open shared object file: No such file or directory
```

After that, I simply set `$LD_LIBRARY_PATH` to `/usr/local/lib/` which solved the problem.

Thanks a lot! :-)

bastibl, 2017-05-08:
great.

closed 2017-05-08 by bastibl