bastibl/gr-ieee802-11 #75: Overflow

cit, 2017-05-10:
Hey,

I'm running `wifi_rx` in gnuradio 3.7.11 with a HackRF and even with 5 Mhz sampling rate, I get a lot of overflows. As far as I understood, overflows is an indicator that the computer is not fast enough to process the receiving data. I tried `wifi_rx` with three different computers (4 x i7-5600U 2.6 GHz, 4 x i5-4570 3.2 GhZ, 8 x 2.7 GHz) and the result was always the same. I did also run `volk_profile` before I started gnuradio. 

Do you have any ideas what might causes the overflow? Or is my only option to get even faster PC. If yes, do you have some minimal requirements for the hardware?

Thanks in advance!

bastibl, 2017-05-11:
You'll probably see a DC offset, which triggers frame detection all the time, completely overloading the receiver. I would start with checking the autocorrelation. See also www.wime-projects.net/installation

cit, 2017-05-11:
Hey,

you were completely right. After I put https://github.com/ghostop14/gr-correctiq after my osmocom Source, I don't get any overflows and the frames are correctly decoded. Thanks again for your great work!

bastibl, 2017-05-12:
great to hear that it worked.

closed 2017-05-12 by bastibl