bastibl/gr-ieee802-11 #77: TX with a hackRF

cit, 2017-05-15:
Dear Bastian,

receiving wireless frames works quite well now. However, now I have problems when sending wireless frame with the example `wifi_tx.grc`. I'm not able to see any frames with another computer that has a wireless card in monitor mode.  

Again, I only replaced the USRP Sink with the osmocom Sink. With the default values, I have a lot of underrun errors in the debug console. The only setup in which I don't get any underruns is with a sample rate of 1 MHz and a sending interval of 10 ms. I tried different channels (2.4 GHz and 5 GHz) without any luck. Do yo have any idea what might be the problem?

bastibl, 2017-05-15:
I assume the problem is not the frame rate (frames per second) but the transfer of the samples to the SDR. With higher sample rates, there might be some hick-ups when the samples are streamed to the device. These interruptions corrupt the frame and there is nothing any receiver can do.
The module creates the whole frame in advance, before starting to stream samples to the SDR (the Packet Pad block is a Tagged Stream Block). So this is really nothing that can be done within the module. It is a problem of GNU Radio/Osmocom Sink/HackRF driver/your USB chip.

cit, 2017-05-15:
I would exclude the USB chip and GNU Radio, because I have tested it with two different computers. One with Ubuntu 16.10 (GNU Radio 3.7.9) and the other with Archlinux (GNU Radio 3.7.11). I'm not sure about the osmocom sink and hackrf. However I'm using the latest firmware for the HackRF. Last weekend, I used the same setup as a FM transmitter to stream music to a radio. This worked without problems. 

Do have a hint for me how I can investigate this problem further?

The output of `wifi_tx.grc` is the following:
```
Executing: /usr/bin/python2 -u /home/cit/Dojo/sdr/wifi_tx/wifi_tx.py

linux; GNU C++ version 6.3.1 20170306; Boost_106300; UHD_003.010.001.001-0-unknown

MAPPER: encoding: 0
set_min_output_buffer on block 9 to 397056
set_min_output_buffer on block 11 to 397056
set_min_output_buffer on block 13 to 397056
set_min_output_buffer on block 14 to 397056
set_min_output_buffer on block 16 to 397056
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in sink types: uhd hackrf bladerf 
Using HackRF One with firmware 2017.02.1 
set_min_output_buffer on block 28 to 96000
set_min_output_buffer on block 31 to 100000
UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU
>>> Done
```

cit, 2017-05-15:
Supplement: I installed the osmocom sink from the git repository (gr-osmosdr v0.1.4-91-gcf954948) without any effect.

I captured the USB traffic that gets transferred to the HackRF. Nearly all URB_BULK_OUT packets are filled with zeros. There is no real data transferred to the HackRF. This is really strange. I compared it with my FM transmitter and there I can see data. If I use a file sink instead of the osmocom sink I can see data as well. Very strange!

[image]

bastibl, 2017-05-16:
Sorry, but I'm afraid I cannot help you much with this problem. I don't have a HackRF and I don't think that the problem is related to this module. I would recommend to post this question to the GNU Radio mailing list. There are probably a lot of people with practical experience with the HackRF.
This module really just creates samples and sends them to the device in one go. If streaming doesn't work, there is nothing I can do.

The 0s that you see are maybe the padding that is added by the Packet Pad block. For some devices this padding is necessary since they need some time to ramp up the TX chain, cluttering the first samples of every frame.

To convince yourself that your problem is really unrelated to this module, you might want to use a message source to create samples and use a PDU to Tagged Stream block to create a sample stream that you can pipe to the SDR.

cit, 2017-05-16:
I did that and I receive underrun errors like before. Nevertheless, thanks for your help! I will write an email to the gnuradio mailing list.

fztfztfztfzt, 2017-07-23:
Hi, I have the same problem with my hackrf. Receiving wireless frames works well with my hackrf. But I have a lot of underrun errors when I run `wifi_tx.grc`. Have you had any solutions about that now? Think you.

cit, 2017-07-24:
Unfortunately no. According to the gnuradio mailinglist the hackrf driver has no support for burst transmissions. They suggested to give [SoapySDR](https://github.com/pothosware/SoapySDR) a try, which is a vendor neutral SDR driver. However, that didn't work for me.

bastibl, 2017-07-24:
It is OK for the HackRF to report one underrun per frame. It doesn't support bursts, doesn't mean that it's not working. After the frame, the sample stream naturally stops and the device cannot know whether that was on purpose or whether the PC didn't produce samples fast enough. So it reports an underrun.

If there are multiple underruns per frame, it means that it stalled during the transmission and corrupted the signal. This would be an actual problem. But since the WiFi module pre-computes all samples before it starts streaming them to the device, there is nothing that can be done here. You probably should check  your HackRF firmware, driver, and USB chip.

As I said, I don't have a HackRF, but I heard that people are using it with this module. So it's probably a config/installation issue (of your hardware driver) that should be discussed at the HackRF repo or the GNU Radio mailing list.

closed 2017-07-24 by bastibl