bastibl/gr-ieee802-11 #78: How to achieve shorter packet interval?

desword, 2017-05-21:
I am going to implement an application that needs shorter packet interval. E.g., emulating the A-MPDU feature that about 28us within two packets to achieve higher throughput.

However, according my experiments, the largest packet transmission frequency is about 183 packet/min
using QPSK 3/4, total packet length 61Bytes, and with interval 100ms.
It first not follows the setting of 100ms packet interval, and then it is far from my requirement.
 So, is there any way to achieve my goal?
Thanks

bastibl, 2017-05-26:
That's a GPP-based SDR. You won't be able to achieve the required accuracy to space the frames by exactly 28us if you just trigger frame transmission every 28us.
You will either have to create a complete sample stream in advance and pad the number of zeros between frames that correspond to 28us to get the exact interarrival time.
If you need something more interactive, you will have to adapt the transceiver to always generate the complete group of A-MPDU frames (again with zeros padded between the frames) before it starts streaming the samples to the device.

desword, 2017-05-27:
Thanks for your reply.
I am new to this kind of details of PHY. Why padding  zeros into a large frame would generate multiple-frames? By the way, what is the maximum PDU length in current implementation? 

If the padding method work and the   maximum PDU length is large enough, I think it is possible to do an accurate packet sending scheduling by inserting multiple padding zeros.
But I am not sure whether there are any problems for this implementation.

Thanks.

bastibl, 2017-05-27:
Maybe that was a misunderstanding. If you randomly insert zeros in a normal frame, it will not transform in a fragmented frame. It would just be corrupted.
What I meant is that you will have to generated the MAC fragments and pad zeros in between them to get accurate timing. Initially, it sounded that you wanted to trigger generation of the fragments with a time delay of 28us and hoped that the frames would be disseminated with that exact timing. This approach won't work.

The current maximum frame size is 1500byte, but it should be straightforward to change through compile time constants.

closed 2017-08-04 by bastibl