bastibl/gr-ieee802-11 #79: Stream frozen after WiFi Sync Long block with LimeSDR on example `wifi_rx.grc`

valer-cara, 2017-08-04:
- OS: Ubuntu 16.04
- Hardware: LimeSDR
- GNU Radio version: 3.7.10
- What are you trying to do: Trying to run the `wifi_rx.grc` example using an `osmocom` source set to the LimeSDR device (`driver=lime,soapy=0`)
- What is you setup: Attempt to dump packets from the local wifi router on Channel 7 (2.442Ghz)
- Bandwidth and frequency:  Channel 7 (2.442Ghz), tried 5/10/20Msps
- What did you already do to debug
  Disabled blocks sequentially to trace back the issue. A QT time sink runs up to the `WiFi Sync Long` block, but is frozen from this block on.
- Where exactly does it break
  When running `wifi_rx.grc`, the stream is frozen
[image]

bastibl, 2017-08-04:
I don't have a LimSDR, but what you see in the screenshot is that the autocorrelation stays high the whole time. This is probably caused by LO leakage/DC offset. As a result, frame detection is triggered all the time, totally overloading the receiver. Try removing this constant from the signal.

Anyhow, this should just result in a total overload and many dropped samples., But not a completely stalled flow graph. There might be a bug, but that I cannot reproduce. Maybe you can enable logging in the sync long block and track it down.

bastibl, 2017-08-09:
Hmm, I was recently also seeing this and other strange issues :-/ Any chances you have a Skylake/Kably lake CPU. If that's the case, can you try disabling hyperthreading in the BIOS.

closed 2017-09-01 by bastibl