bastibl/gr-ieee802-11 #80: cmake .. : -- Configuring incomplete, errors occurred!

alicelovalvo, 2017-09-01:
Hi Sir,
I want to install gr-ieee802-11, but this is the error:

dolphin:build Alice$ cmake ..
-- Build type not specified: defaulting to release.
-- Boost version: 1.59.0
-- Found the following Boost libraries:
--   filesystem
--   system
Checking for GNU Radio Module: RUNTIME
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_RUNTIME_FOUND = TRUE
Checking for GNU Radio Module: DIGITAL
-- Checking for module 'gnuradio-digital'
--   Found gnuradio-digital, version 3.7.11
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-digital.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_DIGITAL_FOUND = TRUE
Checking for GNU Radio Module: FFT
-- Checking for module 'gnuradio-fft'
--   Found gnuradio-fft, version 3.7.11
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-fft.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_FFT_FOUND = TRUE
Checking for GNU Radio Module: FILTER
-- Checking for module 'gnuradio-filter'
--   Found gnuradio-filter, version 3.7.11
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-filter.dylib;/opt/local/lib/libgnuradio-fft.dylib;/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_FILTER_FOUND = TRUE
Checking for GNU Radio Module: PMT
-- Checking for module 'gnuradio-runtime'
--   Found gnuradio-runtime, version 3.7.11
 * INCLUDES=/opt/local/include
 * LIBS=/opt/local/lib/libgnuradio-runtime.dylib;/opt/local/lib/libgnuradio-pmt.dylib
GNURADIO_PMT_FOUND = TRUE
-- Checking for module 'cppunit'
--   Found cppunit, version 1.14.0
--
-- Checking for module SWIG
-- Found SWIG version 3.0.12.
  File "<string>", line 3
    print sysconfig.get_python_lib(plat_specific=True, prefix='')
                  ^
SyntaxError: invalid syntax
CMake Error at /opt/local/share/cmake/gnuradio/GrPython.cmake:113 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  include/ieee802-11/CMakeLists.txt:17 (include)

  File "<string>", line 3
    print sysconfig.get_python_lib(plat_specific=True, prefix='')
                  ^
SyntaxError: invalid syntax
CMake Error at /opt/local/share/cmake/gnuradio/GrPython.cmake:113 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  /opt/local/share/cmake/gnuradio/GrSwig.cmake:25 (include)
  swig/CMakeLists.txt:27 (include)

Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Error at /opt/local/share/cmake/gnuradio/UseSWIG.cmake:199 (add_executable):
  add_executable called with incorrect number of arguments
Call Stack (most recent call first):
  /opt/local/share/cmake/gnuradio/UseSWIG.cmake:256 (SWIG_ADD_SOURCE_TO_MODULE)
  /opt/local/share/cmake/gnuradio/GrSwig.cmake:192 (SWIG_ADD_MODULE)
  swig/CMakeLists.txt:51 (GR_SWIG_MAKE)

You have called ADD_LIBRARY for library _ieee802_11_swig without any source files. This typically indicates a problem with your CMakeLists.txt file
Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
  File "<string>", line 3
    print sysconfig.get_python_lib(plat_specific=True, prefix='')
                  ^
SyntaxError: invalid syntax
CMake Error at /opt/local/share/cmake/gnuradio/GrPython.cmake:113 (file):
  file FILE([TO_CMAKE_PATH|TO_NATIVE_PATH] path result) must be called with
  exactly three arguments.
Call Stack (most recent call first):
  python/CMakeLists.txt:21 (include)

Traceback (most recent call last):
  File "<string>", line 2, in <module>
TypeError: Unicode-objects must be encoded before hashing
CMake Error at /opt/local/share/cmake/gnuradio/GrPython.cmake:125 (add_custom_target):
  add_custom_target cannot create target "ALL" because another target with
  the same name already exists.  The existing target is a custom target
  created in source directory "/Users/Alice/work/gr-ieee802-11/swig".  See
  documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/local/share/cmake/gnuradio/GrPython.cmake:228 (GR_UNIQUE_TARGET)
  python/CMakeLists.txt:29 (GR_PYTHON_INSTALL)

-- Configuring incomplete, errors occurred!
See also "/Users/Alice/work/gr-ieee802-11/build/CMakeFiles/CMakeOutput.log".
See also "/Users/Alice/work/gr-ieee802-11/build/CMakeFiles/CMakeError.log".

Any ideas?
Thank you in advance.

Alice

bastibl, 2017-09-01:
Any chances that this is a leftover from an older installation? I have my cmake files under <prefix>/lib/cmake.

alicelovalvo, 2017-09-05:
Now, everything it-s ok, but when I load some .grc file from this project, this is the output:

** (process:39276): WARNING **: Trying to register gtype 'GMountMountFlags' as enum when in fact it is of type 'GFlags'

** (process:39276): WARNING **: Trying to register gtype 'GDriveStartFlags' as enum when in fact it is of type 'GFlags'

** (process:39276): WARNING **: Trying to register gtype 'GSocketMsgFlags' as enum when in fact it is of type 'GFlags'
<<< Welcome to GNU Radio Companion 3.7.11 >>>

Block paths:
	/opt/local/share/gnuradio/grc/blocks

Loading: "/Users/Alice/gnuradio/gr-ieee802-11/examples/wifi_rx.grc"
Block key "foo_wireshark_connector" not found
Block key "ieee802_11_decode_mac" not found
Block key "ieee802_11_frame_equalizer" not found
Block key "ieee802_11_moving_average_xx" not found
Block key "ieee802_11_moving_average_xx" not found
Block key "ieee802_11_parse_mac" not found
Block key "ieee802_11_sync_long" not found
Block key "ieee802_11_sync_short" not found
Mac OS; Clang version 8.0.0 (clang-800.0.42.1); Boost_105900; UHD_003.010.002.000-MacPorts-Release

I follow you instruction in https://www.wime-project.net/installation/ but the problem is still here.
Thanks in advance.

Alice

bastibl, 2017-09-05:
Citing from this page:

If GNU Radio Companion doesn’t find the blocks it will output error messages like:

>>> Error: Block key "ieee802_11_ofdm_mac" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "foo_packet_pad" not found in Platform - grc(GNU Radio Companion)

In that case you have to adapt your ~/.gnuradio/config.conf to include something like:

[grc]
global_blocks_path = /opt/local/share/gnuradio/grc/blocks
local_blocks_path = ~/usr/share/gnuradio/grc/blocks

alicelovalvo, 2017-09-05:
As I said, I follow your instructions in https://www.wime-project.net/installation/ but the problem is still here

bastibl, 2017-09-05:
It would be very helpful if you could write slightly more verbose bug reports.

Until then, did you blindly copy-paste these paths or did you adopt the local blocks path to the directory where you actually installed the module?

alicelovalvo, 2017-09-05:
I've just solved.. local blocks path was wrong!
Thank you very much for your patience.

Alice

bastibl, 2017-09-05:
Great that it works.

closed 2017-09-05 by bastibl