bastibl/gr-ieee802-11 #81: AttributeError: 'module' object has no attribute 'square_ff'

ayserafettin, 2017-10-04:
Hello everyone,
I am using ieee802.11 transceiver and it works fine but when I try to add a new block in this module it compiles but in runtime it throws this error: " AttributeError: 'module' object has no attribute 'square_ff' " . How can I solve this problem. I please your help 
Thanks sincerely

bastibl, 2017-10-04:
Looks like you try to use gr_modtool and didn't properly fill the boilerplate code that it generates.

Since the question is totally unrelated to this module, but a general question on how to use gr_modtool, it would be great if you could post it to the GNU Radio mailing list.

closed 2017-10-04 by bastibl

ayserafettin, 2017-10-04:
Hi Bastian,
thanks a lot for quick reply. I did send actually but did not get any reply so I thought maybe you or some other friends have some ideas to help me.

Thanks again
Regards
________________________________
Gönderen: Bastian Bloessl <notifications@github.com>
Gönderildi: 4 Ekim 2017 Çarşamba 14:38
Kime: bastibl/gr-ieee802-11
Bilgi: ayserafettin; Author
Konu: Re: [bastibl/gr-ieee802-11] AttributeError: 'module' object has no attribute 'square_ff' (#81)

Looks like you try to use gr_modtool and didn't properly fill the boilerplate code that it generates.

Since the question is totally unrelated to this module, but a general question on how to use gr_modtool, it would be great if you could post it to the GNU Radio mailing list.

bastibl, 2017-10-04:
If you didn't plan to create a square_ff block, just search through the code to see where you will have to change stuff.

If you actually wanted to create such module, it's hard to tell where things went wrong. The error you posted means that python doesn't know about the block. So the swig binding might be worth a look.

But really, there are 1000 things that can go wrong. Easiest might be to just post your module on GitHub and ask the list to have a look.

ayserafettin, 2017-10-04:
Actually I have another block, which I implemented giving the same error. To to be sure I created the block by following the instruction from Gnuradio example. It is the same code that they give in the example. So it gives error. It might a python error as you mentioned but if then how can I solve it?how can I change the python path?If python can see the block in your receiver why it cannot see the block that i created inside gr-ieee802.11 module?
Thanks
________________________________
Gönderen: Bastian Bloessl <notifications@github.com>
Gönderildi: 4 Ekim 2017 Çarşamba 15:19
Kime: bastibl/gr-ieee802-11
Bilgi: ayserafettin; Author
Konu: Re: [bastibl/gr-ieee802-11] AttributeError: 'module' object has no attribute 'square_ff' (#81)

If you didn't plan to create a square_ff block, just search through the code to see where you will have to change stuff.

If you actually wanted to create such module, it's hard to tell where things went wrong. The error you posted means that python doesn't know about the block. So the swig binding might be worth a look.

But really, there are 1000 things that can go wrong. Easiest might be to just post your module on GitHub and ask the list to have a look.

bastibl, 2017-10-04:
The list of available blocks in GRC is generated based on XML files. It is completely independent from actually existing or working blocks. You have a valid XML file, but some Python problem.

Again, I recommend uploading the module and posting to the list.