bastibl/gr-ieee802-11 #83: Issue using wifi_rx.grc

steven-bloom, 2017-10-19:
Hallo, Bastian. Thanks for this wonderfull tool! 
However, Id like to ask you few questions (problems). 
Like many of users i had replaced UHD-Device with Hackrf in order to sniffing Wi-Fi. 
[image]
But i see nothing in Wireshark in realtime or in File-Sink
[image]

I changed the Preferences (Sample-Rate, Freq, gain) but still can not see anything.  

What can be the problem?
I really appreciate any help you can provide

bastibl, 2017-10-19:
I just ordered a HackRF yesterday to test it myself.

What you see is a huge DC offset. This triggers frame detection all the time. You will have to compensate it. I'll have to test, which exact parameters for the DC blocker work.

steven-bloom, 2017-10-19:
Thank you very much for your quick response and advise! With interest and curiosity I will wait for news from you :)

bastibl, 2017-10-20:
You're welcome. You don't necessarily have to wait. It should be very easy for you to try out. 
Use a Single Pole IIR Filter to filter the IQ stream from the SDR and subtract it from another copy of the IQ stream from the SDR.
The filter has only a single parameter, alpha. Add a slider to vary this value and start the flow graph. Play with different values for alpha and see what works.

bastibl, 2017-10-24:
I've received my HackRF and it worked as described in my previous post. Also, I found the DC offset to be very different on different channels. It's not ready to get pushed, but I don't see a big issue for getting it to work with the HackRF.

steven-bloom, 2017-10-25:
Hallo, Bastian!
Thank you for your responce. I'm sorry,I think I did not quite correctly understand you. Is that block correct established?

[image]

bastibl, 2017-10-25:
The IIR filter calculates an average, so now you average out the signal, leaving you only with the DC offset. That why I said that you have to subtract this from the signal. Anyhow, maybe this is just to complicated. I'll  push some changes in the next weeks.

steven-bloom, 2017-11-15:
Hello Bastian, sorry for inconvenience.
I was wondering about your experience with HackRF + gr-ieee802.11. Could you transmit and receive  Wi-Fi Signal? 
What concerns me is that i could compensate this DC-Peak, nevertheless it did not improved SNR and i 
 could not Tx/Rx the WI-Fi Signal. I think, that the problem lies in the device itself. Because when start, i see "U" - Symbols flowing  in the debug window. Do you have the same?
[image]
[image]

Yours faithfully, Steven.

SUDIPTOBORDOLOI, 2017-11-15:
Hi Bastian,
 I am facing a problem when i execute the flowgraph. It opens and its closes immediately. Ans as a result it doesnt capture any packets. What do you think could be the problem. Your gr-802.15.4 Zigbee RX works fine in the same system. 

Hope the system stats help you to answer the question,
GNU Radio Companion 3.7.9
uhd-host 3.9.7
Ubuntu 16.04lts
Intel Core i5 with 4GB RAM
USRP B205mini-i

..Program Log

gr::log :INFO: sync_long0 - LONG: frame start at 320

>>> Done (return code -11)

Generating: '/home/sdr/gr-ieee802-11/examples/wifi_rx.py'

Executing: /usr/bin/python -u /home/sdr/gr-ieee802-11/examples/wifi_rx.py

linux; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_003.009.007-release

-- Detected Device: B205mini
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 16.000000 MHz... 
-- Actually got clock rate 16.000000 MHz.
-- Performing timer loopback test... pass
-- Setting master clock rate selection to 'automatic'.
-- Asking for clock rate 20.000000 MHz... 
-- Actually got clock rate 20.000000 MHz.
-- Performing timer loopback test... pass
Using Volk machine: sse4_2_32_orc
Ogr::log :INFO: sync_long0 - LONG: frame start at 320
Ogr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 229
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320
gr::log :INFO: sync_long0 - LONG: frame start at 320

>>> Done (return code -11)

bastibl, 2017-11-15:
@SUDIPTOBORDOLOI please create a separate issue (with more context).
@steven-bloom please use a channel in the 5GHz band as they are usually less crowded. Also the sample rate at sender and receiver have to match. You send a 5MHz signal, but the receiver expects 10MHz. There should be at most one 'U' per frame. Otherwise, it indicates that the PC does not send the samples fast enough to the SDR, which corrupts the signal.

steven-bloom, 2017-11-17:
Unfortunately, this did not help. I still get these running U-Characters when transmiting

bastibl, 2017-11-18:
From what you write it is hard to tell what's your problem. As I said, if there is only one U per frame, it might be OK. If there are multiple Us, the signal is corrupted.
I'm not sure what you are currently trying to do, but to understand if your problem is RX or TX, consider testing it with normal WiFi cards on the 5GHz band.

albertobrincat, 2018-01-19:
Hello Bastian, i'm using your tool with hackrf one, i've replaced only UHD source with osmocom one, in order to have hackrf compatibility.
I'm using the same configuration that i saw on this discussion but i saw multiple "O" on output.
If you tested it with hackrf, could you please tell me witch configuration you have used?
Thanks in advice
[image]

bastibl, 2018-01-19:
The 'O's only indicate that your PC cannot keep up with the incoming samples and has to drop them. This can have many reasons (very old PC, runs inside VM, USB issues, no SIMD/VOLK, DC offset, and many many more). I'm not sure what configuration you are referring to. I didn't do anything special. Please see this and the other threads for more information. Let me know if you have a more specific question.

ynyn11121, 2018-02-05:
Hello Bastian, i'm using your tool with a usrp b210, a wifi card, a laptop and a desktop
I  connect  wifi card with my laptop, connect usrp b210 with my desktop,  by using wifi_rx.grc ,I  cannot  receive wifi signals,  I don not know how to configure  my wifi card .Thank you so much .

bastibl, 2018-02-05:
There are many ways to generate WiFi frames. The easiest one might be to open an 11a AP in the 5GHz band and receive the beacons.
AFAIS, your problem has nothing to do with this issue. Please open a separate issue if you have any other questions.

mohammad-shekarifard, 2018-02-13:
Hello Bastian, I want to use the wifi_rx.grc and then see the output packets in the wireshark, but in the wifi_rx.grc the wireshark block is disable and i can not connect wireshark to the gnuradio, can you please help me ?. Thank you.

bastibl, 2018-02-13:
Is that question in any way related to this issue? Just enable the block through the menu bar or press 'e'

mohammad-shekarifard, 2018-02-13:
[image]

bastibl, 2018-02-13:
You have to follow the installation instruction and actually install gr-foo. The block is missing, i.e., not installed.

https://www.wime-project.net/installation/

As I already said, this has nothing to do with this issue. If you have a problem, please open a new one.

bastibl, 2018-03-01:
OK, so I was playing a bit around with the HackRF. I used the filter design tool of GNU Radio to design a high-pass filter to filter out the DC offset. It's by no means perfect but worked reasonably well for me.
The flow graph is available here:

https://github.com/bastibl/instant-gnuradio/blob/wlpc/assets/wifi_rx.grc

closed 2018-03-01 by bastibl

newtosdr, 2018-04-28:
@bastibl Hello Bastian,  I have used your fiow graph [](https://github.com/bastibl/instant-gnuradio/blob/wlpc/assets/wifi_rx.grc) and run with hackrf.  I modified the Sample rate from 5M to 10M, and the feq from 5.89G to 2.452G. Why can't I capture standard wifi packets? I'm quite sure that there are many wifi signals around.

bastibl, 2018-04-29:
Not all signals on 2.4GHz are 11g. I'd recommend to test it with 11a on 5GHz first.
In any case, 11g and 11a use 20MHz channels, so you have to set the sample rate to 20MHz,

newtosdr, 2018-05-02:
Thanks, Bastian! I have a question about 11a, how could you generate the 11a signal on 5GHz?

bastibl, 2018-05-02:
You configure a sample rate of 20MHz and tune to a channel in the 5GHz band. There are buttons for both options in the user interface.

newtosdr, 2018-05-02:
I mean what is your source of 11a signal? Is it generatde by another hackrf or came from wifi signals around? I just modified the parameters as you said and tried it again, but didn't get anything. It was very confusing.

bastibl, 2018-05-02:
That doesn't matter, but I would recommend to just use any WLAN card to send 11a frames and receive it with the SDR. Since you have some issues with your setup, I would use the simpler setup with one SDR to avoid adding unnecessary complexity.

bastibl, 2018-07-21:
Did you configure the file sink to repeat the file. So that it is sent over and over again?

YuexinZhao, 2018-07-22:
This  is the modified flow graph in wifi_rx.grc. I configured the file source not to repeat the raw data file and set the file sink overwirte file and unbuffered off. 
OS: Ubuntu 14.04 ;   Gnuradio: 3.7.9 
[image]

bastibl, 2018-07-22:
can you upload the raw file somewhere?

YuexinZhao, 2018-07-22:
@bastibl Below is a Dropbox link of the raw file :)
[https://www.dropbox.com/s/yxqtc6b3i54zm4l/Data_20180722.rar?dl=0](url)

YuexinZhao, 2018-07-23:
@bastibl  I have tried a Throttle block with different rate, including 1MHz, 500kHz and 100kHz, but the problem still exists! Could you test my raw data and grc file? Any help will be appreciated！

bastibl, 2018-07-23:
This was tricky. The reason for the changing results are the Moving Average blocks. They don't compute the average of the whole window for every sample but maintain a sum that is updated for every sample. The new sample is added and the one that drops out of the window is subtracted. That's more efficient then summing up the whole window for every sample.

The drawback of this approach is that there might be tiny deviations from real sum, since floating point operations are not perfect. To avoid that these deviations add up, the sum is recalculated every time the block is run by the GNU Radio scheduler. This is not deterministic, so there might be very small differences.

Your trace contains quite a lot low-SNR ACKs, where these small variations make a difference. If you use FIR filters (as shown in the screenshot) you don't have these variations (but experience higher CPU load).

[image]

YuexinZhao, 2018-07-23:
@bastibl  The troubled problem has been solved based on your modifications. Thank you very much for your detailed explanation and warm help :-)

Pradyumna007, 2020-02-01:
Hi, 

I am getting an error saying "error can not create connection". Could you please tell me how to resolve this?

[image]

bastibl, 2020-02-02:
You replaced the UHD source with a Lime SDR source and in the process probably made some wrong connections. Revert to the original flowgraph and make sure you connect it similarly.

MarlonBedoya, 2020-02-10:
Hello Bastian. Is it possible to use it in "monitor mode"? I need to construct something like the typical applications for scanning all the SSID in all the channels, similar to the wifi-analyzer for android or LinSSID for Linux. I can scan it on Wireshark using the card network, but I can't get it with the USRP.

bastibl, 2020-02-11:
This is a PHY implementation. It is always in monitor mode. It receives and forwards all frames. It is not associated to an AP/SSID or whatever. If you don't receive frames, it's not an issue with monitor mode. Check frequency, gain, sample rate, if the antenna is connected to the correct port, etc etc

MarlonBedoya, 2020-02-12:
Yes. I can receive frames and decode it, choosing in the GUI the channel number. What I mean is the possibility of receiving and decoding frames from all the channels and decode the channel number as well, for example. It is not just from a particular channel, but also from the whole wi-fi spectrum. 
I can do it on Wireshark by activating the monitor mode option, it being disconnected from the network, and it listens to all the channels and decode parameters like your implementation, but including the channel number.

bastibl, 2020-02-12:
It is not possible to receive on all channels at once. But you can, for example, change the channel periodically. This can be done programmatically by posting a command to the USRP source: https://www.gnuradio.org/doc/doxygen/page_uhd.html

Pradyumna007, 2020-02-13:
@bastibl I have built it again. But still I am facing the same problem. Could you please tell me how to resolve this? I have built gr-foo first and there were no errors. Later I have built gr-ieee802-11. There were no errors during the installation. But when I am opening, I am seeing these blocks disabled.
[image]

bastibl, 2020-02-13:
The screenshot looks perfectly fine. If that's the unmodified flowgraph, there should be no error creating connections.
If you want to log received packets to disk, just enable the grayed-out blocks through the context menu, the menubar, or by pressing'e'.

Pradyumna007, 2020-02-13:
Thank you. That worked.

MarlonBedoya, 2020-05-11:
Hello. I have used the flowgraph with several modifications for individual purposes, and everything is going well so far. But now, I need to move the most I can towards FPGA because I need to liberate CPU for other processes ( At least the correlation for sync segment). I haven't found information or examples about how I should do it... Can you explain to me How I can migrate it to FPGA?

bastibl, 2020-05-11:
An FPGA implementation for WiFi is beyond the scope of this project. It is not trivial to move to FPGA and it depends on the HW you use. Please see OpenWiFi for an Open Source FPGA implementation for WiFi: https://github.com/open-sdr/openwifi

MarlonBedoya, 2020-05-11:
I mean, It is possible to use the Spartan 3 what is **inside the USRP N210** to migrate a segment of the process? I am thinking about modifying the framework to have the correlation output before the wifi sync short block, but I don't know if my thoughts whether or not reachable.

bastibl, 2020-05-11:
I know what you mean. It is possible, but it is hard. And this is not the place to ask for help with FPGA implementations on the N210. If you really want to go down that road, please ask on the USRP-users mailing list.

elizabethrivera, 2020-09-03:
Hello Bastian @bastibl
Thank you so much for your excellent contribution. I recentely implemented the wifi_rx.grc, but all the packets in the wireshark
are marked as Malformed Packets. Do you have any idea of the possible issue?

bastibl, 2020-09-07:
What do you mean. All frames that make it to Wireshark were decoded correctly (i.e., the CRC was correct). What exactly is your setup? You probably send test frames that are valid WLAN frames but have arbitrary higher-layer payload. The WLAN loopback flowgraph, just sends a punch of 'X's, for example. This is displayed as malformed, but only on higher layers. All WLAN headers should be fine.

Paula-res, 2021-04-05:
I try the wifi transceiver flowgraph with the x310 I failed to get constellation plots I just get noise.