bastibl/gr-ieee802-11 #87: About the wifi_rx

mingleiyou1988, 2017-11-03:
Hi bastibl,

I am trying to embed the wifi_rx into a larger program, but I am running into a trouble. It would be really great if you can provide some suggestions on it.

The code is implemented using N210. The whole program is divided by two parts, the first part is to use fft to scan the channel, while the second part is to use wifi_rx/tx to communicate with each other. Since fft and wifi_rx reuse the same hardware resource, during  every iteration the following sequence is executed in the first part:

lock wifi_rx => start fft_block => stop fft_block =>unlock wifi_rx

While during the second part:

check if the wifi_rx received anything (an indicator variable is added in the wifi_rx receiver function) 

But I find that during the second part of the main iteration, the wifi_rx cannot pick all the frames from the wifi_tx with the program I write. I have also tried the standard wifi_tx, and change the interval of the tx packages, it is the same that some frames are not received. With shorter interval, the frames can be picked are increased. Also the rate of success or not seems to be random. I have also used another USRP to monitor the tx while running the pair, and the tx frames can be received by standard wifi_rx.

My major question is, what is the condition for the wifi_rx to capture a frame? The CPU has to be available? Any suggestions or comments would be sincerely appreciated!

Thanks,

Ming

bastibl, 2017-11-07:
I'm not sure if I understand your problem, but it seems the lock-reconfigure-unlock cycle introduces too much delay. I assume that also the SDR has to stop and restart streaming every 1ms.
To avoid this, you could just run both branches (the FFT and the normal receiver) in parallel in one flow graph and send PMTs with your indicator variable to the WiFi receiver. This can be done from another GNU Radio block or from somewhere else in your application.

mingleiyou1988, 2017-11-07:
I think you have got the point! I have tested that if the FFT is avoided during the first part, the wifi_rx can work totally fine. I have also tested that the gtk interface and RX parameters are not the main reasons for the problem. Hence your analysis is the best reason so far. 

I will try your suggestions to parallel the FFT with RX in one flow graph, but it might take a while to do this. I will close this issue for now and let you know if it works (fingure cross) or not later.

mingleiyou1988, 2017-11-28:
Hi bastibl,

I regrouped the blocks as suggested in my code, and I am trying to use the gr-foo.selector to dynamically switch between FFT route and wifi_rx route. But I find that when I call gr-foo.selector.set_output_index(output_index), the whole program stuck at the self.unlock() within the sub function set_output_index. 

I try to use a dummy simple grc with a gui_chooser to choose the output index, and the switching seems fine, although the wifi rx cannot pick anything.

Do you have any suggestions on this issue?

Thank you in advance!

Ming

bastibl, 2017-11-28:
I'm not sure what you are doing. What I suggested was to always run both branches (no lock-unlock) and async messages between the branches. The selector from gr-foo doesn't solve your problem, because, under the hood, it's doing exactly that, i.e., is reconnecting the flow graph in a lock-unlock cycle.

Apart from that, your problem sounds like one of the blocks is in a busy loop and the scheduler cannot jump in an lock the flow graph. An easy way to check is to use htop and enable thread names.

mingleiyou1988, 2017-11-29:
Thank you for the suggestions! I am trying to run them at the same time. Will update the details later!

mingleiyou1988, 2017-12-10:
I have tried to implement the fft block and rx block sharing the same usrp source. It works with test cases, but not in the full program. During the fft procedure, the system is overwhelmed by 'D' warning.

I have debugged the program for two weeks and find that it seems the problem appears when the fft, wifi_rx, wifi_tx run at the same time. Even if I initialise the wifi_tx but not start it, the 'D' warnings appear during running. If the wifi_tx block is not initialised at all, there are no 'D' warnings.  

My guess is the computing ability for my laptop is not enough to support all of them running at the same time. I will try the whole thing on a more powerful pc later.

Besides the hardware limitations, do you have any suggestions on the implementation of wifi_rx and wifi_tx at the same computer end? Is there a way to roughly estimate how much burden some blocks may bring to the whole program?

bastibl, 2017-12-10:
A 'D' indicates a network problem. The problem might, therefore, not be related to CPU power, but a network switch or a bogus configuration when two devices are initialized. I guess the current problem is not directly related to the module, but an issue of you UHD driver, setup, or configuration.

mingleiyou1988, 2017-12-10:
Thanks for the prompt reply! 

The final goal of the program is a cognitive radio with one computer and one N210. Hence all blocks, such as fft, rx, tx are running at the same computer. The laptop and N210 is directly connected, so it rules out the switcher problem. I will dig in the other options. 

Thanks again!!

mingleiyou1988, 2017-12-13:
Hi bastibl,

I have tried to use htop to monitor the running, and it gives some interesting clue.

The system is stable without tx blocks. When I run the rx and fft, the peak time for the system looks like this.
[image]

Things only go wrong when tx blocks are added. When I run the rx and fft with tx, and if the system is not overwhelmed by the 'D' packages, the peak time for the system looks like this.
[image]

When I run the rx and fft with tx, and if the system is overwhelmed by the 'D' packages, the system looks like this.
[image]

It tells me a few things when .

1. In the test case, the tx blocks are initialized and started, but the message sender is never called. Yet the mapper associated to the tx blocks is really active according to the CPU usage.

2. When the system is overwhelmed by the 'D' packages, it is doing FFT, which is shifting the central frequency of the usrp source. During this period, the syn-short and syn-long are very active according to the CPU usage. 

3. During FFT, before the  'D' packages appears, I can observe that the CPU usage of syn-short and syn-long build up until it finally gives the 'D' packages.  

I am guessing that the active of the mapper in tx block and syn-short, syn-long have roles in the 'D' package issues. Based on these observations, could you please give some suggestions?

bastibl, 2017-12-13:
I'm not sure what you mean by 'calling the sender'. Are you starting the TX flow graph, but not the periodic message source? Or did you do any other modifications? It looks like something unexpected happens in your TX.

While you first should fix the TX, the second question is how you operate RX and TX. I guess on the same frequency. But do you use full-duplex or half-duplex?

Also, I'm not sure what you are trying to send. It looks a bit like a continuous stream with frames back-to-back.

mingleiyou1988, 2017-12-13:
Regarding the TX flow graph:
I have deleted the periodic message source and instead I use a message_sender block with a message out port using message_port_register_out method. This port is connected to the 'app in' port of ieee802_11_mac_0 block using the msg_connect method. This modification allows me to send messages only when it is needed. Similarly, I have defined a message_receiver block, which set a flag to indicate a message has been received, while in the main loop the flag is checked to read the received messages (if there is any). 

In the test case, the FFT, RX and TX blocks are started, but the message_sender is not called. As it is the only message source, the TX blocks should not be active at all, but as can be seen in the pictures of the previous reply, the mapper, which is the message port of 'app in' that the message_sender connected to, is very active. So even if I don't post anything on the port of 'app in' of mapper, it is still working?

Regarding the operation of RX and TX:   
The RX and TX is running at the same frequency, but I am using them in a half-duplex way. The ideal case is that node 1 send => node 2 receive => node 2 response => node 1 receive. But now I just let one node to talk while the other only listen. The EM leakage on the daughter board is an issue, I have tried  the iq balance and dc offset for both tx and rx, which has suppressed this leakage somehow, which is indicated by a better constellation while receiving. 

Regarding what is sending:
I am using an ADC board to do some measurement, and in the main loop I send this measurement to another node. The frequency to send the measurement is controlled to avoid flooding the TX blocks, which is around 500ms a frame. Only occassionly I can see 'U' packages.

Update on the debugging:
I have increased the sensitivity of the syn-short block from default 0.56 to 0.7, and the 'D' packages disappeared. Maybe the usrp source samples during frequency changing has triggered the syn-short block too much? But what has triggered the mapper block? I think I still need to seek your suggestions on these questions.

Thank you veryveryvery much for your time and kind help!!

bastibl, 2017-12-13:
Frame detection works based on the auto-correlation of the signal. When there is any leakage, that will trigger frame detection all the time. With increasing the sensitivity, you might have changed the threshold so that leakage doesn't trigger frame detection anymore. 
You could try to get rid of this with offset tuning (see Ettus manual), close the enclosure, or place sender and receiver farther apart from each other.

Regarding the TX, use a Message Debug block to debug the PMTs that are put into the receiver. I guess something is wrong there. Make sure this are not too many and their format matches the one of the Periodic Message Source.

mingleiyou1988, 2017-12-13:
Thanks for the prompt reply!

Regarding the tx block:

To debug the tx, I have tried the default wifi_tx block you provided, and it seems the mapper is still very busy.

[image]
 
Is this normal? In the grc there is a block with ID ieee802_11_mapper_0 while on the grc it shows WiFi Mapper, but in htop it seems this is the one with the name mapper6. The number at the end may vary, at least from my observation.

Regarding the rx block:

Indeed the change of sensitivity is changing the threshold, is there a way to dynamically change this parameter?

Thanks,

Ming

bastibl, 2017-12-13:
Depends, if you didn't change the parameter to send much larger frames and/or at a much higher rate then there is something wrong. Can you please try the loopback flow graph (without any modifications).
If there is a problem, I would suggest you use clean checkout and rebuild the module.

Regarding RX, no there is currently no way to adjust the sensitivity dynamically. You would have to extend the code. But really what you want is get rid of the leakage (offset tuning, more space between devices, DC offset correction, ...).

mingleiyou1988, 2017-12-14:
I have tried the loopback grc without any modifications, and the mapper performs in a similar way.
[image].

May I just double-check that this is not supposed to be the case? I have tried this on three different laptops and the performance of the mapper block is exactly the same. If it is possible, could you please have a look at the loopback grc and htop on your side? It would be really helpful for me to get an idea on the performance I should expect from this block. 

Regarding the RX part, I have tried all the mensioned methods to suppress the leakage, but it seems not helpful to the problem. Maybe during the frequency changing the sync-short and sync-long may experience sparks, since the samples during frequency tuning are not stable as they are supposed to be?  Another guess is the mapper has stressened the CPU which leads to the congestion? These two guesses can be both supported by the fact that problems only appears when tx-blocks are added.

bastibl, 2017-12-14:
I have no issues. Can you please try the next branches of GNU Radio and this module. There have been changes to the GNU Radio scheduler that might explain your issue. (In prior release of GNU Radio, some types of block had to do busy waiting for messages).

mingleiyou1988, 2017-12-17:
In short: 
It is indeed the GNURadio scheduler problem that has caused the wifi_mapper block busy problem. With the 'NEXT' branch of GNURadio, the problem seems to be solved, at least for the wifi_mapper part.

I have tried several versions and it seems that only the next branch has updated the scheduler. But the next branch of GNU has some big changes in QTGUI features, one of them I believe is on the pyqt5. I have tried some combinations of qt/pyqt/pyqtw dependencies and they all fail to enable this feature. With some combinations it passes the cmake but fails during make test and crashes after installation. For now the machine just runs without qtgui, but luckily I am using gtk instead of the fancy qt codes. I think this is not the right place for this installation issue, but it would be really great if you still would like to provide some suggestions on the dependencies for the qtgui feature.

Thank you once again for your kind help on the many useful suggestions as well as your time on reading and responding to my questions!! Hope you have a nice Christmas holiday!

Best,

Ming

P.S. 
As you are also a core contributor of the GNU Radio, a summary of the prolem might be of some help with the next version of the GNU radio.

The problem:
Some modules are very busy during running. The one appears to be troubled in my case is the wifi_mapper block, as can be indicated in from the screen shots above. The CPU usage is very high and the running time accumulates along with the program running. This block seems never rest. Note that although this busy block occupies almost one CPU core all the time, the whole program (or .grc ) is still able to work with expected results. So without observing the threads performance using tools like htop, one might not notice this problem.

Known GNU versions possibly with this problem:
I was running 3.7.10.1-211-gb49da16 when the problem appeared, which was installed using the build-gnuradio script a long time ago. 

I have tried some other versions: the master branch with version 3.7.12git-295-ga0adcd3. 

The version that 'solved' the problem: 
The next branch of GNU raido with version 3.7.12git-1261-g3c63f73. The updates of for this version includes the improvement of GNURadio scheduler you described. The same program gives the following result.

[image]

bastibl, 2017-12-18:
Glad that this solved your problem. I never had problems installing the next branch. If you have any issues, you could write to the GNU Radio mailing list, but usually, the output from cmake should tell you, what libraries you are missing. The next branch of this module should work with the next branch of GNU Radio.

Regarding the changes of the scheduler: we are aware of the issues of the master branch. However, overcoming the busy waiting required changes to the blocks that are not backward compatible. Therefore, the fix is only in next and not in master.

mingleiyou1988, 2017-12-18:
It is always a pain to deal with the dependencies, I will see what I can do. My program runs gtk instead of gt, so it is not a problem to me, yet. Hope the next version will be released soon! 

Thank you once again for your help!

closed 2017-12-18 by mingleiyou1988