bastibl/gr-ieee802-11 #88: Asynchronous buffer overflow, dropping message

omarsalaheldine, 2017-11-25:
I am currently using the gr-ieee802 code to transmit over a channel emulator using 2 USRPs X310.
When trying to send pdus at slow data rates (500 kbps) the transmitter and reciever work correctly and I receive the messages.
However, when I increase the datarate to 2 Mbps, it works correctly for a short duration then I recieve the asynchronous buffer overflow message ans by checking the received rate at the receiver node, it hits around 600 kbps at max.

I am using 5Ghz sample rate on both ends and center frequency of 1Ghz without any interferences (using channel emulator) I tried all supported modulation schemes and the data rate increases by around 50 kbps from scheme to another.

Using the benchmarks, I got higher datarates than 2 mbps from the X310

Am I missing something ?

bastibl, 2017-11-25:
I assume you mean 5MHz. Dependent on the MCS and packet size, a data rate of 2Mbps might not be physically possible. You will have to do the math to calculate what can get through. The fact that different phy leat to a different data rate doesn't tell you much.
That being said, the transmitter is not designed to high data rates. See [this paper](http://www.ccs-labs.org/bib/arcos2016accelerating/) for a discussion of the performance. If your plan is to send frames back to back to in measurements, you will first have to generate a file with a given inter-frame space (zero padding per frame) and then send the file. Otherwise, the inter-frame space varies due to the non-determinism of GNU Radio scheduling.
Currently, you are trying to send the messages too fast and since there is no back-pressure mechanism for asynchronous messages, GNU Radio starts dropping messages.

omarsalaheldine, 2017-11-26:
Yes I am sorry, I meant 5MHz. Thank you for your response and the attached paper really gives much information regarding the issue. I was able to increase the data rates by applying some of the acceleration methods suggested. 
Thanks again.

closed 2017-11-26 by omarsalaheldine