bastibl/gr-ieee802-11 #93: Does this project support other boards?

authuir, 2018-01-17:
Will this project works well on other boards? Or are there any lists of supported-hardware?

bastibl, 2018-01-18:
The project does not rely on hardware-specific features and, therefore, works with SDRs that are supported by GNU Radio and that operate on the frequency band of interest, offering a bandwidth that is large enough for the signal that is supposed to be studied.

authuir, 2018-01-23:
Thank you for your reply. 
But I have another question: some hardwares support multi-antenna which means it can run MIMO system like 802.11n , but some hardwares like Hackrf only support half-duplex , and I wanna ask that will those hardware-specific features affect ?

bastibl, 2018-01-23:
> I wanna ask that will those hardware-specific features affect?

I'm not sure if I understand you correctly. There is no impact on this module. Even if the device has many antenna ports, the transceiver will not automatically switch to MIMO or transform into an 11n transceiver. So there is no difference if you use an SDR with one or more antenna ports. If you want to do MIMO, you would have to extend the code.

closed 2018-01-23 by bastibl