bastibl/gr-ieee802-11 #95: Problems with simulation of wifi_loopback.grc

emozo, 2018-01-24:
Hello Bastian, thanks for your work. I have installed the ieee802.11 package and I can see the decoded symbols in the constellation, but I think that I have two problems.
 1.-The  error rate always remains at zero for any combining of modulation-codification.
  2.- I can not see any package in the wirechark, which I installed and enabled in the wifi_loopback.grc block.

Thank you, and I appreciate any suggestion about it.

bastibl, 2018-01-25:
The error rate indicator is not a precise measurement, but a simple estimate. Once nothing gets through anymore, it will also not be updated anymore. Still, for me, it displays errors. I would suggest you try again with 16-QAM and play a bit around with the parameters.
The Wireshark blocks are (AFAIK) disabled by default. Enable them in GRC and open the generated file in Wireshark.

emozo, 2018-01-27:
Thank you, I did it and it was ok. Regards

closed 2018-01-27 by emozo