bastibl/gr-ieee802-11 #96: bastibl/gr-foo#6 - Extended Explanation

lfasanchez, 2018-01-25:
Hello Bastian,
It's a pleasure to write to you, regarding your work con SDR for 802.11.

I am trying to install the gr-foo software, but a message comes through:

[  8%] Building CXX object lib/CMakeFiles/gnuradio-foo.dir/channel_model_impl.cc.o
In file included from /home/cecad/Wifi-GNURadio/ieee802-11/GR-foo/gr-foo/lib/channel_model_impl.cc:18:0:
/home/cecad/Wifi-GNURadio/ieee802-11/GR-foo/gr-foo/lib/channel_model_impl.h:27:47: fatal error: gnuradio/filter/mmse_resampler_cc.h: No such file or directory
compilation terminated.

I have read the explanation on:

gr-foo :Installation error #85
and on:
bastibl/gr-foo#6

Nevertheless, i am new in this and i don't understand how can i make the branches  correspond _(" you have to use the next branch of gr-foo with the next branch of GNU Radio and the master branch of gr-foo with the master branch of GNU Radio´.)_

Would you be so kind, explaining me how can i do this and the asociated commands for achieving this.

Once again thanks for the help.

bastibl, 2018-01-25:
If I understand you correctly, your question is about installing software in general and/or about using git. In a nutshell, you have to checkout the branch in git (see the [website](https://git-scm.com/) for more information) and install GNU Radio (see their [Wiki](https://wiki.gnuradio.org/index.php/Main_Page) for more information).
Let me know if you have any questions specific to this module.

lfasanchez, 2018-01-27:
Hello Bastian, 
Thanks for the quick answer, i had issues understanding the branches, but once i updated  everything it worked, thanks again.

bastibl, 2018-01-27:
Great.

closed 2018-01-27 by bastibl