bastibl/gr-ieee802-11 #97: Issue with undefined symbol (libgnuradio-ieee802_11.so)

augustojv, 2018-01-30:
Hi,

I'm facing the following problem (which happens for both 'master' and 'next' branches):

```
$ python
Python 2.7.12 (default, Nov 20 2017, 18:23:56)
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import ieee802_11
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/__init__.py", line 37, in <module>
    from ieee802_11_swig import *
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 28, in <module>
    _ieee802_11_swig = swig_import_helper()
  File "/usr/local/lib/python2.7/dist-packages/ieee802_11/ieee802_11_swig.py", line 24, in swig_import_helper
    _mod = imp.load_module('_ieee802_11_swig', fp, pathname, description)
ImportError: /usr/local/lib/libgnuradio-ieee802_11.so: undefined symbol: _ZN2gr5blockC2ERKSsN5boost10shared_ptrINS_12io_signatureEEES6_
>>>
```

The symbol in question is:

```
$ c++filt _ZN2gr5blockC2ERKSsN5boost10shared_ptrINS_12io_signatureEEES6_
gr::block::block(std::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, boost::shared_ptr<gr::io_signature>, boost::shared_ptr<gr::io_signature>)
```

My environment is:

- Ubuntu 16.04
- GNURadio 3.7.9 (installed from Ubuntu's repo)
- gcc/g++ 4.9.3
- LD_LIBRARY_PATH=/usr/lib:/usr/lib/x86_64-linux-gnu/:/usr/local/lib/
- PYTHONPATH=/usr/local/lib/python2.7/dist-packages

I'll appreciate any advice on this issue, since I've spent days trying to fix it without any success.

Thanks in advance.
Augusto

bastibl, 2018-01-31:
This is strange. If there is a problem with this symbol, it should already show up during compile time. Make sure to use the same environment variable during compilation and execution. Also, check that the GNU Radio libraries are actually installed in the folders that you set as LD_LIBRARY_PATH.
Maybe also try `sudo ldconfig`.

augustojv, 2018-01-31:
Bastian, thanks -- I just found the problem. It was about a discrepancy in the version of gcc/g++. The GNU Radio package available in Ubuntu 16.04 is compiled with gcc/g++ 5.3. At some point in the past, I had to downgrade it to gcc/g++ 4.9, which was the version I used to compile your library. Now I went back to gcc/g++ 5.3 and everything seems to work fine.

Thanks.
Augusto

closed 2018-02-05 by bastibl