bastibl/gr-ieee802-11 #98: Issue with receiving  wifi signal

ynyn11121, 2018-02-05:
Hello Bastian, i'm using your tool with a usrp b210, a wifi card, a laptop and a desktop
I connect wifi card with my laptop, connect usrp b210 with my desktop, by using wifi_rx.grc ,I cannot receive wifi signals, I don not know how to configure my wifi card .

I use a router to set my wifi signal is  802.11g  11 channel, then use my wifi card to surf the internet ,but my usrp still cannot catch good wifi signals.

ynyn11121, 2018-02-05:
These pictures can help tell what I want ro say,
[image]
[image]
[image]

bastibl, 2018-02-05:
I'm sorry but these pictures are not too helpful. I'm not sure what you are trying to communicate with them. What you are saying is "I tried; it didn't work". It would be helpful if you could be a bit more specific.

ynyn11121, 2018-02-05:
Sorry，my English is not good，so i just want to express  by picture.
What I want  to say is : 
I use a router to set my wifi signal is 802.11g and 11 channel 2.462Ghz, but my usrp still cannot catch good wifi signals by using wifi_rx.grc.

ynyn11121, 2018-02-05:
Thank you so much, I  just study about SDR  for 3days, so  many technical details are not very clear for me.

ynyn11121, 2018-02-05:
My experiment result is ,I do not know what my problem is.
[image]

bastibl, 2018-02-05:
Looks like you modified the flow graph, so it's hard to tell what's going wrong. I'd recommend you use the unmodified flow graph from the repository (you changed the sample rate. Unfortunately, the receiver does not support 40MHz/11n. It's more then just setting the sample rate to 40Mhz....).
To avoid potential problems, I'd recommend you open the AP in the 5GHz band. On 2.4GHz, the card might operate in a backward compatible mode that is not supported by the module.

closed 2018-02-13 by bastibl